motioneye-project/motioneye #2624: Project dependencies may have API risk issues

PyDeps, 2022-10-26:
Hi, In **motioneye**, inappropriate dependency versioning constraints can cause risks.

Below are the dependencies and version constraints that the project is using

```
jinja2
pillow
pycurl
tornado
```

The version constraint **==** will introduce the risk of dependency conflicts because the scope of dependencies is too strict.
The version constraint **No Upper Bound** and **\*** will introduce the risk of the missing API Error because the latest version of the dependencies may remove some APIs.

After further analysis, in this project,
The version constraint of dependency **jinja2** can be changed to *>=2.7,<=3.1.2*.
The version constraint of dependency **pillow** can be changed to *==9.2.0*.
The version constraint of dependency **pillow** can be changed to *>=2.0.0,<=9.1.1*.
The version constraint of dependency **tornado** can be changed to *>=3.0,<=6.2b2*.

The above modification suggestions can reduce the dependency conflicts as much as possible, 
and introduce the latest version as much as possible without calling Error in the projects.

The invocation of the current project includes all the following methods.

<details><summary>The calling methods from the jinja2</summary>
 <pre>jinja2.Environment.get_template
</pre>
</details>
<details><summary>The calling methods from the pillow</summary>
 <pre>PIL.Image.new
PIL.Image.open
</pre>
</details>
<details><summary>The calling methods from the tornado</summary>
 <pre>tornado.ioloop.IOLoop.instance
tornado.httpclient.AsyncHTTPClient.fetch
tornado.httpclient.HTTPRequest
tornado.iostream.IOStream.__init__
tornado.httpclient.AsyncHTTPClient.configure
tornado.iostream.IOStream.connect
tornado.iostream.IOStream
tornado.httpclient.AsyncHTTPClient
</pre>
</details>
<details><summary>The calling methods from the all methods</summary>
 <pre>PIL.ImageDraw.Draw
tornado.ioloop.IOLoop.instance.start
multiprocessing.Pool.apply_async
now.weekday.split
int_or_0.split
os.readlink.read
int.replace
shlex.split.lower
self._get_folder_id_by_name
header.lower.startswith
s.unicode.encode
self.write
decoded.split.lower
MjpgClient.clients.get
str.count
media_list.sort
datetime.datetime.fromtimestamp
call_hook
s.hashlib.sha1.hexdigest
re.compile.sub
tornado.ioloop.IOLoop.instance.add_timeout
utils.urlopen
base64.encodestring
data.get.lower
[... 730 lines cut ...]
out.strip.strip
f.write
c.get
utils.make_str
share.replace.replace
email.Utils.formatdate
main
logging.getLogger.removeHandler
MjpgClient
network_share.lower
list.add
_get_time_zone_md5
motionctl.get_motion_detection
get.save
self._error
re.sub.strip
locals
filtered_camera_ids.append
</pre>
</details>

@developer
Could please help me check this issue?
May I pull a request to fix it?
Thank you very much.

coder0107git, 2022-10-26:
@PyDeps After doing some looking I found that those are the old motion eye dependencies (`python2` branch). Also, I found that the `dev` branch (the new python3 version) [dependencies](https://github.com/motioneye-project/motioneye/blob/8bd8486c8ed1315f36a3581277b5178db0b0e1ce/setup.cfg#L24-L30) has the same dependency version issue you mentioned. I think something like: 
```
install_requires =
    tornado == 6.2.0
    jinja2 >=2.7,<=3.1.2.
    pillow == 9.2.0
    pycurl == 7.45.1
    babel == 2.10.2
    boto3 == 1.25.1
```
should work.

JamBalaya56562, 2026-11-11:
Closing, as the dependencies are left open on purpose: `pyproject.toml` sets a lower bound only where motionEye needs a newer release (`tornado>=6.5.8`), and CI installs the latest releases on Python 3.9 to 3.15 for every pull request, so an incompatible release shows up there.

closed 2026-11-11 by JamBalaya56562

MichaIng, 2026-11-11:
In case of Tornado, a certain newer version is needed recently, indeed, but the main reason I added the lower boundary there is dependabot vulnerability scans. If we do receive a report by that infrastructure, we set/raise the lower boundary. Though the alerts may not happen if there is no lower boundary set at all.

An upper boundary is discouraged in any case, as this would prevent latest security patches to be applied on install, and break compatibility with other projects who do set lower boundaries. It would make sense for a lock file, to generate reproducible environments and document a working state, but not in `pyproject.toml`.