motioneye-project/motioneye #2625: Google Drive is no longer functional in Settings, File Storage, Upload Media Files

starbasessd, 2022-10-29:
See ticket in motionEyeOS: https://github.com/motioneye-project/motioneyeos/issues/2954

LittleWenlock, 2022-10-31:
See # #2611. I did point out the warning doing Obtain Key "Google will soon block apps that don’t comply with Google’s security policies ...this app should stop using unsupported OAuth flows". 

I guess that has now happened.

xjon685, 2022-11-08:
Hi
Here is the link to the OOB migration https://developers.google.com/identity/protocols/oauth2/resources/oob-migration

So it looks like all uploads to google drive now fail.   Do the devs have any plans to fix this or maybe integrate something else like nextcloud or backblaze B2 https://www.backblaze.com/b2/docs/ ?

Daniel-Pivonka, 2022-11-13:
I am also facing this issue. Here is my workaround for now for others looking for a solution. 

I originally ran motioneyeos but switched to running rasbian and installing motioneye. This way I could run a different tool that supports the new auth method to sync to google drive. So I dont use the "upload media" option in motioneye instead I'm using [rclone](https://rclone.org/).

I followed this guide to [install and setup rclone](https://medium.com/@artur.klauser/mounting-google-drive-on-raspberry-pi-f5002c7095c2) besides the last couple steps about mounting your drive. I didnt need my full drive fully and permanently mounted. I then setup a cron job to run every 15 mins and use the [rclone sync](https://rclone.org/commands/rclone_sync/) command to sync my cameras media directory to a folder on my google drive.

`*/15 * * * * /usr/bin/rclone sync /var/lib/motioneye/Camera1/ gdrive:/cam/ --exclude lastsnap.jpg --exclude *.thumb --config /home/pi/.config/rclone/rclone.conf --verbose 2>> /tmp/rclone_cron.log >> /tmp/rclone_cron.log`

This copies the files saved from "Camera1" to a folder on my google drive called "cam", excludes thumb nail files that are also saved in that directory. Also it saves a log in /tmp.

stefansundin, 2024-10-23:
It is possible to work around this problem. I looked through a few issues related to this and I didn't seen anyone post this solution, so hopefully this will help a few people.

The only issue is that Google no longer allows the OOB redirect URI that motioneye uses. So we can just replace that with e.g. `https://localhost/` and then manually copy the authorization code. We have to modify the motioneye code though, so that the application uses a valid URI when exchanging the authorization code for credentials.

We need to update these two lines:

https://github.com/motioneye-project/motioneye/blob/72af3711d28175669ca06ab09913b788a5fd8617/motioneye/uploadservices.py#L170

https://github.com/motioneye-project/motioneye/blob/72af3711d28175669ca06ab09913b788a5fd8617/motioneye/uploadservices.py#L280

To instead be:

```python
            'redirect_uri': 'https://localhost/',
```

Anyway, enough talk, here's how you do it:

```shell
# I installed motioneye using pip, your installation directory may differ.
# Here's how I find my directory:

$ pip show motioneye
Name: motioneye
Version: 0.43.0
Summary: motioneye, a multilingual web interface for motion.
Home-page: https://github.com/motioneye-project/motioneye
Author: Calin Crisan, Jean Michault, ...
Author-email: author@example.com
License: GNU General Public License v3.0
Location: /usr/local/lib/python3.9/dist-packages
Requires: pillow, babel, pycurl, boto3, jinja2, tornado
Required-by:

# So my installation directory is /usr/local/lib/python3.9/dist-packages
# Navigate to it:

$ cd /usr/local/lib/python3.9/dist-packages

# Modify uploadservices.py using an editor that you are comfortable with:
$ sudo vim uploadservices.py

# Edit the two lines as mentioned above, replace urn:ietf:wg:oauth:2.0:oob with https://localhost/
# Make sure you save the file!!

# Then restart motioneye (IMPORTANT):
$ sudo systemctl restart motioneye
```

If you did the steps above correctly, you should now be able to perform the authorization without error. Once you do that, you will be redirected to a non-working URL, e.g. `https://localhost/?code=4/gkzLKg4b25v59dqesjma5KpNy42v2zyaMHuvpU3tVtADocDYTtjzcEkZCMWqiigVQRAmXou&scope=https://www.googleapis.com/auth/drive.file`

You need to take the `code` in that URL and paste it into the `Authorization key` text field in motioneye. In this example my code would be `4/gkzLKg4b25v59dqesjma5KpNy42v2zyaMHuvpU3tVtADocDYTtjzcEkZCMWqiigVQRAmXou`

Click the `Test service` button and it should work.

Hope that helps. Enjoy! 😄

sammmuel, 2024-12-30:
Thanks a lot @stefansundin, it's a so cool workaround! 🥇
👍

AndyMAu, 2025-01-03:
Thanks,I will give it a try. My authorization key to upload to Google Drive lasted until mid November 2024.

sammmuel, 2025-01-04:
hello,
It seems to me that it's no longer possible to upload video from several cameras to the same Google drive directory, but I did manage to upload video from several cameras to different directories on my Google drive.

ArgiesDario, 2025-01-21:
I tried @stefansundin's workaround with my MotionEye running on Docker, and I'm sharing it helps

Open a shell in the docker container (assuming its name is motioneye):
`docker exec -it motioneye /bin/bash`

Find the path of `uploadservices.py`:
`find / -name "uploadservices.py"`

Replace the file (assuming the path is the default one):
`sed -i "s|urn:ietf:wg:oauth:2.0:oob|https://localhost/|g" /usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py`

Check if the change was succesful:
`cat /usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py | grep "redirect"`

Exit the container and restart MotionEye:
`docker restart motioneye`

Unfortunately, even if all changes were made correctly, this workaround is not working for me. The non-working URL I receive does not include the `code` query parameter to copy.

I managed to make it work using [rclone](https://github.com/rclone/rclone) by syncing the folder where MotionEye stores the videos with Google Drive.

JamBalaya56562, 2026-11-03:
Fixed by #3512: Google blocked the out-of-band redirect that "Obtain key" used, and motionEye now uses a loopback redirect (`http://127.0.0.1`) with PKCE. After allowing access, the browser ends at a page that does not load; paste its whole address as the authorization key. Checked with a Google account: key obtained, folder created, and a file uploaded to Drive. The fix is in 0.45.0b1 on PyPI (`pip install --pre --upgrade motioneye`) and will be in v0.45.0, so the `uploadservices.py` workaround above is no longer needed. Closing as completed; please reopen if Google Drive uploads still fail with it.

closed 2026-11-03 by JamBalaya56562