motioneye-project/motioneye #2626: Struggling to setup MotionEye

Junaid0411, 2022-10-30:
I am using:
Buster OS
RPi4B 2GB

This is my current screen:
[image]

I believe it should be working, however, when I try to connect to the IP it says 'refused to connect'

Any help would be greatly appreciated! :)

MichaIng, 2022-10-31:
From the logs it looks all good. motionEye is listening on the expected port?
```sh
ss -tlpn
```

Junaid0411, 2022-11-01:
[image]

I'm currently using WiFi, I'll try Ethernet and see if I get it working tok

JamBalaya56562, 2026-11-11:
Closing, as this got no follow-up after MichaIng's question. motionEye listens on all interfaces on port 8765 by default (`listen 0.0.0.0`, `port 8765` in `motioneye.conf`), so `http://<ip>:8765/` with the port is the address to open. Please open a new issue with `ss -tlpn` and the log if that is still refused with a current release.

closed 2026-11-11 by JamBalaya56562