motioneye-project/motioneye #2629: WARNING: Connect error on fd 16: ECONNREFUSED

aidanjacobson, 2022-11-03:
Hi, I am using motioneye 0.18.0 on an odroid xu4 as an addon (docker container) through hassos. It was working fine, until I started getting this error. A complete uninstall/reinstall of motioneye did not solve the problem, so while the error seems to be generated by motion, the problem seems to lie elsewhere. What gives?

MichaIng, 2022-11-04:
It seems the container was crashing/halting. However, this is an issue on container level. Might it be an issue with insufficient RAM or other system limitations?

Issue has been reported two times on the HA addon's repo, but stated to have been fixed:
- https://github.com/hassio-addons/addon-motioneye/issues/115
- https://github.com/hassio-addons/addon-motioneye/issues/120

The guys over there will habe better insights about which underlying issue can trigger such errrr.

JamBalaya56562, 2026-11-11:
Closing, as MichaIng found the container itself stopping, which belongs to the add-on (hassio-addons/addon-motioneye#115 and #120, reported as fixed there). Please open a new issue with the log if it still happens with a current add-on.

closed 2026-11-11 by JamBalaya56562