motioneye-project/motioneye #2631: RTSP Stream Support?

nickadian, 2022-11-09:
Good afternoon!

I am trying to send out an rtsp stream from motion eye but don't see the option. I know the option is available through motioneyeos . What is the best way to stream out via rtsp from motioneye?

BrendanGrant, 2022-11-12:
I came here to ask the same thing.

I followed the setup instructions to install motionEye on top of an Ubuntu 22.04.1 Live Server, and I'm limited to the following two camera types:

* Remote motionEye Camera
* Simple MJPG Camera

If I plug in my RTSP URL for either, I am told "Protocol "rtsp" not supported or disabled in libcurl"

shooftie, 2023-10-08:
All but one of the cameras i have run MotionEyeOS v0.42.1 (Motion v4.2.2+gitUNKNOWN) and have the option to stream via RTSP. 

[image]

I recently installed MotionEye v0.43.0 (Motion v4.5.1) onto a DietPi machine and that option is not available. I assumed, because it was not an OS install.

[image]

I have been using MotionEye for years, albeit without much tinkering, this is now become a problem as I am unable to find any RTSP stream.

My approach moving forward is to install DietPi on the camera machines and then MotionEye on that, thus regaining some of the more traditional Linux behaviour that MotionEyeOS excludes but maintaining the downright helpful configuration that makes MotionEye worthwhile.