motioneye-project/motioneye #2634: high usage of motioneye processor on raspberry pi 3 model b

steinhorstbr, 2022-11-12:
doing some tests on raspberry with raspian + motioneye the processing is at 100 on the 4 cores.

The stream is rtsp, and I configured motioneye to record 9fps, however, even deactivating the recording, the processor is at 100% of use, it is only visualized

What could be happening?

aaronouthier, 2022-11-13:
Try this: go to a command prompt. Type `ffmpeg -hwaccels` . If “omx” is on the list that is returned, then add `netcam_decoder omx` in the empty box under the web ui in video device->extra options.

It might or might not help.

steinhorstbr, 2022-11-14:
the result of the command was

`root@raspberrypi:/home/pi# ffmpeg -hwaccels
ffmpeg version git-2022-11-12-bfab87a Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --enable-gpl --enable-nonfree --enable-mmal --enable-omx --enable-omx-rpi
  libavutil      57. 42.100 / 57. 42.100
  libavcodec     59. 52.101 / 59. 52.101
  libavformat    59. 34.101 / 59. 34.101
  libavdevice    59.  8.101 / 59.  8.101
  libavfilter     8. 50.100 /  8. 50.100
  libswscale      6.  8.112 /  6.  8.112
  libswresample   4.  9.100 /  4.  9.100
  libpostproc    56.  7.100 / 56.  7.100
Hardware acceleration methods:
vdpau
vaapi`

aaronouthier, 2022-11-14:
Check if the device `/dev/dri/renderD128` exists?
If so, add `netcam_decoder vaapi` to extra options.
Reboot, and check CPU usage.
If that doesn’t help, or the device doesn’t exist, instead try `vdpau` in place of vaapi .

steinhorstbr, 2022-11-14:
does not own.
/dev/dri/renderD128

vdpau
I did and still with 100% cpu

aaronouthier, 2022-11-14:
So, a permissions issue. Try adding the user that MotionEye runs as to the video group.

So, for example, if the user is named “motion”, then:
`sudo usermod -aG video motion`
You can skip the sudo keyword if already logged-in as root.

If that fails for some reason, find the group that owns /dev/dri/renderD128 and substitute that for ‘video‘ above.

aaronouthier, 2022-11-14:
Update: on my system, the group is ‘render’, not video.

So, in my case, I would use
`sudo usermod -aG render aaron`

aaronouthier, 2022-11-22:
A few questions for you. How did you install MotionEye?
What version are you running?

JamBalaya56562, 2026-11-11:
Closing, as aaronouthier's questions got no reply. With an RTSP camera, most of the CPU goes to Motion decoding the H.264 stream in software, so a lower-resolution substream as the camera URL helps the most; the hardware decoding aaronouthier described works only where ffmpeg has a usable decoder for the device.

closed 2026-11-11 by JamBalaya56562