motioneye-project/motioneye #2637: parallel video recording

Tormund1985, 2022-11-15:
I have configured the motion eye to detect on the one hand, photos on motion detections, as well as small videos with these detections.

However, at the same time I would like to record the complete video for 24 hours.

How could I do it through motioneye? If I select continuous recording I would lose the videos of the detections, and I want to keep both.

Many thanks.

zagrim, 2022-11-15:
This had also been briefly discussed in #1954 where another solution was proposed, but basically since that is not directly supported by Motion, there's no support for it in ME, either. In any case it requires having "something else" recording the continuous video, be it the camera itself recording to some cloud storage, or having another box as MotionEye hub as was proposed in the issue I linked, or something like you came up with yourself.

I guess all the options have some clumsiness, and one thing to mind might be how to catch if the continuous recording fails at some point, and if it could be restarted sooner than at the next midnight. But that of course depends on the case, it might be ok to fail until the next midnight.

Tormund1985, 2022-11-16:
> This had also been briefly discussed in #1954 where another solution was proposed, but basically since that is not directly supported by Motion, there's no support for it in ME, either. In any case [...]

Many thanks, I will work to try obtain an efficient solution because with ffmpeg I am not giving good results without re-encoding.

Regards

clavay, 2023-01-23:
You can define a new network camera in MotionEye using the address http://localhost:8084

JamBalaya56562, 2026-11-11:
Closing, as zagrim answered this: Motion records a camera either on motion or continuously, not both, so the full recording needs something next to it, like your ffmpeg segment command, or a second camera in motionEye on the same stream (`http://localhost:8084`), as clavay suggested.

closed 2026-11-11 by JamBalaya56562