motioneye-project/motioneye #2638: Way to hide or remove logo, motionEye, and Calin's name?

abacefbeca, 2022-11-18:
The person I want to hide the camera from has access to the network and could try to visit the camera's IP address where they would find a motionEye logo and the names. Is there a way to remove those parts so that the things seen at the login menu won't help identify the purpose of the device or possibly disable the web interface entirely if necessary? Maybe I could re-enable the web interface if needed by using SSH?

TazerReloaded, 2022-11-20:
You could just firewall the web interface port, for example with nftables. If you need access again you can temporarily disable that firewall rule. A simple nftables config would be:

```nftables
table inet filter {
    chain input {
        type filter hook input priority 0; policy accept;
        tcp dport 80 counter reject comment "Reject connections to motioneye";
        counter comment "Counter for accepted packets";
    }
}
```

zagrim, 2022-11-21:
@abacefbeca I'm sorry if I am mistaken of your purpose behind this request, but to me this sounds like you are spying/stalking on somebody and trying to hide that from your target person. There might be good and legal reasons for that, but in most cases that's not something I consider ethical.

abacefbeca, 2022-11-21:
@zagrim For what it is worth from a stranger on the Internet, normally I wouldn't do something like this and I wouldn't set up a hidden camera to outright spy on or stalk someone. In general terms, I have a situation where the person in question has been unplugging a security camera creating a zone that isn't covered by a camera. The objective would be to gather evidence of what, if anything, is being done while this person thinks they are off-camera. Of course, there are more details. I considered sharing them privately though it seems Github has no means to privately message someone and perhaps the details are of little consequence to you. I really don't know. My apologies if my request isn't appropriate. We can close this if that is what is preferred.

zagrim, 2022-11-22:
@abacefbeca my point merely was to voice my concern that what you described might as well be a abuse case. Apologies if that is not what you are up to, but like you said we are just strangers on the Internet without any history that would speak for our character or intentions. Technically inclined people, like the most of us here offering solutions for issues, tend to be focused on how to make things happen and less on if it should be done at all, and personally I feel that the ethics are pretty important.

Anyway, enough of that I think.

What @TazerReloaded proposed is one option (and if you have `iptables` instead of `nftables` on the box that runs ME, rejecting traffic to the web UI port can be done with that, too, with guides all over the Web), and most likely one of the simplest ones. ME itself doesn't have any option to disable the web UI.

abacefbeca, 2022-11-22:
Thank you. Yes, I looked in to TazerReloaded's suggestion but found that by default MotionEyeOS, or at least the version I have set up on my Pi, has iptables rather than nftables. After a bit of Googling I found some instructions that I think will work to flip the switch, so to speak, for allowing or blocking access to the web interface. I am somewhat familiar with Linux but not super familiar with it. Initially I had no idea what program might be used to block access to the web interface and @TazerReloaded TazerReloaded gave me a hint for where to start looking.

Personally, I think the issue should be dealt with differently but I'm not the one making that call. It isn't only me and this other person involved in the situation. Still, this hidden camera option might not be feasible for other reasons. I guess at this point I am looking in to options.

closed 2022-11-22 by abacefbeca