motioneye-project/motioneye #2640: Invalid motion conversion specifiers

jowax, 2022-11-20:
Hi, I've noticed that the generated `camera-XX.conf` have have invalid conversion specifiers.

For example the following config is generated in my `camera-1.conf` file.
```
...
on_event_start /root/motioneye/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %%t
on_event_end /root/motioneye/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %%t
on_movie_end /root/motioneye/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %%t %%f
on_picture_save /root/motioneye/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %%t %%f
...
```

This results in the following callback generated by `relayevent.sh`
```
http://127.0.0.1:8765/_relay_event/?_username=admin&event=start&motion_camera_id=%t&_signature=...
```
Which in turn results in a 403 in the backend
```
 WARNING: 403 POST /_relay_event/?_username=admin&event=start&motion_camera_id=%t&_signature=... (127.0.0.1) 77.29ms
 ```

According to the motion [documentation](https://motion-project.github.io/4.4.0/motion_config.html#conversion_specifiers) the `%%t` and `%%f` should have a single percent sign instead of two.

MichaIng, 2022-11-20:
I think I found it. When converting from `%`-formatted string to f-string, the `%` escaping of the `%` character itself became obsolete.

MichaIng, 2022-11-20:
PR up: #2641
Would be great if you could verify that it produces correct and functional releyevent script entries now.

jowax, 2022-11-20:
Hi, thanks for the fast response.
Just tried the changes and seems to work as expected!

Here's the new generated config
```
...
on_event_start /root/motioneye_2/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
on_event_end /root/motioneye_2/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
on_movie_end /root/motioneye_2/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
on_picture_save /root/motioneye_2/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
...
```

closed 2022-11-20 by MichaIng