motioneye-project/motioneye #2645: Contribution

DiagnozHub, 2022-11-21:
Hello, i wanna to fix one small bug in css, how can I contribute ?

s17534, 2023-01-13:
Hey, how can I clone 0.43.0 dev git repo? From what I see, there is available 0.42.1 version and installation of mentioned 0.43.0 version is only happening via method:

`python3 -m pip install --user 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
`

In such case, there is no github repo information in above package:

[image]

MichaIng, 2023-01-13:
With `pip` you do not clone repos but install Python modules. If you want to clone the repo, use `git`:
```sh
git clone https://github.com/motioneye-project/motioneye
git fetch --all
git checkout dev
```

s17534, 2023-01-13:
Thank you, that is precisely what I needed.

JamBalaya56562, 2026-11-11:
Closing, as MichaIng answered this: fork the repository, commit on top of the `dev` branch and open a pull request, as s17534 then did.

closed 2026-11-11 by JamBalaya56562