motioneye-project/motioneye #2646: Raspberry Pi 4b - Bullseye, high processor usage and slow video streams

shadmanhabib, 2022-11-22:
_Disclaimer: This is my first post here so apologies if information isn't given in the best structure._

Hey guys, I've got some issues with motionEye and general quality of life issues. I think a lot of it may be due to my setup but would be great to get some advice.

### Usage
Running motionEye to record video streams from 5 network cameras to an external SSD. These streams are passed onto Home Assistant which manages all my smart home devices. Home Assistant also streams all my devices to HomeKit but that's not relevant here. 

Together this forms the smart ecosystem and surveillance system at my house.

### Operating Environment
My setup is as follows:
- Raspberry Pi 4B running Bullseye Lite.
- 8GB ram with 332GB microSD card.
- 2TB SSD connected with an SSD to USB 3.0 adapter (used for storing video files).

### Brief Camera Settings
- All cameras running at 1080p (or similar).
- RTSP streams from the cameras are at 25fps but video device frame rate on motionEye is set to 5fps.
- Video streaming frame rate in motionEye is set to 5fps as well.
- Movie recording is active with the movie passthrough option turned on to reduce CPU load.
- Motion detection in motionEye is on for 2 cameras.

### Issues
#### High CPU usage and low RAM usage
CPU usage on all 4 cores are generally pinned at or close to 100%. Given that the Pi only needs to read from the RTSP cameras and write that information directly to the SSD without any extra encoding because of movie passthrough being turned on, I don't expect the CPU to be at full load all the time.

RAM usage is very low at around 700MB out of an available 8GB. I would expect this value to be higher when dealing with video streams. Initially I thought it was because the `/tmp` folder wasn't set up as `tmpfs` and so some of the slow behaviour (next issue) was because of constant writes to the microSD card. I changed this with the below code but did not observe any measurable difference.

```
sudo cp /usr/share/systemd/tmp.mount /etc/systemd/system/tmp.mount
sudo systemctl enable tmp.mount
sudo systemctl start tmp.mount
```

Here is a screenshot of the `htop` output.

[image]

The Raspberry Pi does have a cooling fan so it's not overheating and throttling. Although that's probably not too useful to know in this case.

#### Slow/Delayed RTSP streams from cameras
High CPU usage wouldn't bother me if everything else worked fine. But this is the bigger problem. I'm getting up to 60 seconds of delay between an event happening and that being reflected in the video stream from motionEye. This is actually pretty bad if I'm trying to use one of the video streams as a video doorbell...the person doesn't load into frame until they've left.

Loading the RTSP video streams from the cameras directly in VLC or similar gives me between 1-2 seconds of delay. I expect the Raspberry Pi to be a bit slower than this but 60 seconds is a huge difference.

### Solution?
Any help is greatly appreciated. At the moment I genuinely don't know why it behaves this way. Is this possibly a Bullseye issue and the new V4L2 drivers that aren't ready yet? Does hardware acceleration need to be manually set up?

Thanks in advance to anyone that took the time to read this.

earlywaff, 2022-11-23:
The problem plays out on many devices and on many systems.
Confirm it on both pi4 and intel 5095 pc

earlywaff, 2022-11-23:
I'm sorry, it's just that English is not very good.
I have checked several devices, I have not found a solution.
Searching through the branches I saw that a new motioneye development is underway and I have to wait.

earlywaff, 2022-11-23:
https://github.com/motioneye-project/motioneye/issues/2634
Maybe this will help you.

shadmanhabib, 2022-11-23:
From https://github.com/motioneye-project/motioneye/issues/2634#issuecomment-1313788103 I attempted `netcam_decoder vaapi` to enable hardware acceleration. My CPU usage is still pinned at 100% on all 4 cores and the video streams seem to be glitching more.

I'll keep these setting for a few days and update on performance. At the moment it seems like waiting for the next release is the easiest option lol.

MichaIng, 2022-11-23:
Isn't VAAPI an Intel CPU feature, expectedly not available on ARM/RPi? https://forums.raspberrypi.com/viewtopic.php?t=321033

shadmanhabib, 2022-11-23:
Yeah my quick google search beforehand told me that too. Just thought I must've been missing something.

Honestly just very new to and hence very confused by drivers in general...FKMS and V4L2 and all.

zagrim, 2022-11-23:
> Given that the Pi only needs to read from the RTSP cameras and write that information directly to the SSD without any extra encoding because of movie passthrough being turned on, I don't expect the CPU to be at full load all the time.

But you also said:
> Motion detection in motionEye is on for 2 cameras.

Just reading the stream and saving to storage should not be an issue for CPU, that's true, but doing motion detection  is, and RPi isn't that powerful. Are you using `netcam_hires` to do motion detection on a smaller resolution stream and keep on saving the high-res stream? You might benefit from that if your cameras offer low-res streams.

Myself I have RPi3 with 2 cams with motion detection, and even with doing motion detection on low-res streams (704x576, 4fps), the system runs quite hot during warmer seasons (passive cooling with the "official" heatsinks, semi-open case). It was not possible at all to do motion detection on the full resolution (currently it is 1280x960 but I think I tried also somewhat lower options without much success).

zagrim, 2022-11-24:
@Tibladar I think that would be generally a good advice, but in OPs case here I wonder whether it actually helps or not. I have not used `movie_extpipe` myself and could thus be wrong here, but isn't that only used when saving movies without the use of `movie_passthrough`? At least [`movie_passthrough`](https://motion-project.github.io/motion_config.html#movie_passthrough) should "create movie files of the motion with the packets obtained directly from the camera", while [`movie_extpipe_use`](https://motion-project.github.io/motion_config.html#movie_extpipe_use) in the docs is said to specify "whether to send the output to a pipe for external encoding into a movie." I'm not familiar with how exactly Motion processes the passed-through video stream, though... at least it must select which frames are saved, but can it do that without actually going through some encode-decode loop?

shadmanhabib, 2022-11-24:
@zagrim I tried turning off motion detection on the two cameras that had it turned on and I'm not seeing a big difference in processor utilisation or stream delay. All cores are hovering between 90-100% with the stream delay still close to 60 seconds.
[image]

shadmanhabib, 2022-11-24:
My recorded videos generally seem fine and smooth at 25fps and I think this is thanks to movie passthrough being turned on.

What's strange is that after a restart the stream delay is short at around 3-5 seconds but then with time it seems to get longer and longer. Any ideas on why that might be the case?

zagrim, 2022-11-25:
@shadmanhabib I suspect that the increasing delay is at least partly due to the high CPU usage, but there might also be some other (incompatibility?) issues behind that. You are not the first one to report that, but I don't know of any specific solution (or the actual reason) for that, and I've myself also experienced that, although it was another software before tying out ME.

Which versions of MotionEye and Motion are you currently running? Since the last `htop` output shows `python2` process I guess you haven't yet tried the current dev branch of ME as [suggested above](https://github.com/motioneye-project/motioneye/issues/2646#issuecomment-1324851953)? That uses Python 3 and has support for Motion 4.4.

shadmanhabib, 2022-11-25:
@zagrim yes I'm on motionEye V0.42.1 and Motion V4.3.2

Do you know if Motion V4.4 has a big performance improvement?

zagrim, 2022-11-25:
I have absolutely no idea, but your issue seems more like a bug of some sort, or lack of support for some edge case or codec flavour or something like that, since there's not really any good reason for such a high CPU usage if motion detection is not being used. So, in lack of any better ideas, upgrading seems worth trying. And Python 2 is anyway dead and buried so there's that, too, even though that shouldn't have any effect on this issue.

shadmanhabib, 2022-11-27:
Update: Upon closer inspection it seemed to only be the doorbell camera that had the long delay. All the others seemed to be under 3 seconds generally. I'm not sure if this changed after turning motion detection off or if it was always like this and I just didn't notice since I was focused on the doorbell camera.

I've adjusted some of the camera and video stream settings for that camera. The video stream is generated with a Raspberry Pi CM4 2GB through the [rtsp-simple-server](https://github.com/aler9/rtsp-simple-server) application by aler9.

Initial impressions seem to be a shorter delay but will test/update over the next couple of days.

shadmanhabib, 2022-11-30:
After adjusting settings on the doorbell camera, I'm still seeing a slowly increasing delay on it so I think for now at least I can isolate the issue to that specific stream.

Question about motion detection on streams that have movie passthrough turned on. How does that work? Does it mean a rolling shutter is kept or frames to record before a motion event? Does that possibly have a big impact on CPU utilisation?

hoangnamitc, 2022-12-31:
My same , I am using motioneye on docker and my CPU alway 95 - 100% . continuously like that 
and I don't have any method to fix this yet. Temporarily I have to stop this container to avoid the noise from the fan, and I'm still looking for a way to fix it,

shadmanhabib, 2023-08-06:
I’ve moved over to a full server PC running the edge version.

No delays on that system. Seems like Raspberry Pi’s just don’t have the processor grunt :(

closed 2023-08-06 by shadmanhabib