motioneye-project/motioneye #2650: Video recording action button(s)

evandene, 2022-11-24:
can somebody help me with this one.
My question(s) are at the bottom of this message.
I run MotionEye on an RPI zero 2W and and on an BPI M2 Zero and all runs very smooth.
I have created two buttons at the right bottom of the screen, "light_on_1" and "light_off_1" who do Start Motion Detection" and "Stop Motion Detection". See picture.
In that way I can manually start and stop movie recording. All works very well. No problem at all.
The script "light_on_1" is as follows:
#!/bin/sh
curl "http://localhost:7999/1/config/set?emulate_motion=1"
The script "light_off_1" is as follows:
#!/bin/sh
curl "http://localhost:7999/1/config/set?emulate_motion=0"
Please note: Motioneye does not allow the scripts having a file extention, so just "light_on_1" and "light_off_1". The files do need to be in /etc/motioneye and made executable with chmod +x light_on_1 and the same with the other file.
Question:
Would it be possible for the development team for:

- Offering Recording ON icon including action
- Offering Recording OFF icon including action

PS: the below command line input does the job also:
wget -O- "http://<hostname>:7999/1/detection/start" >/dev/null
wget -O- "http://<hostname>:7999/1/detection/pause" >/dev/null

zagrim, 2022-11-25:
Requested sooo many times already and #229 is tracking this.

MichaIng, 2026-10-19:
The funny thing is that our action API supports `record_start` and `record_stop`, but nothing is done in the backend behind it. #2691 adds eventstart/eventend actions, which get just forwarded to the motion control API, as a very similar workaround to manually trigger a recording.

What is actually the practical difference between those and `emulate_motion=1`, if any? The `action` API does not have `webcontrol_parms` restrictions. As "action", it seems more explicit than changing a config value. But I do still wonder.

closed 2026-10-20 by MichaIng

MichaIng, 2026-10-30:
Action buttons for toggling motion detection and a motion event manually have been added with #3468. Manually triggering a recording only, without e.g. event notification, is difficult, since `motion` does not support such action, but is blocking the camera device, so a manual FFmpeg call our end cannot capture the video stream either. But starting/stopping a manual motion event, with motion-triggered video recording on, comes close.