motioneye-project/motioneye #2652: Rights to write file (recordings or stills)

speeskeek, 2022-11-25:
hello.

I upgraded to the pyton3 version a while ago. working fine, running stable.
but - recording (using motion detection) is not working since motioneye cant write.

probably a simple permission thing, but dont know what user/rights to give the folder.

`[1:ml1:Terras] [ERR] [ALL] myfopen: Error opening file /media/usb-disk2/CCTV/Terras/2022-11-25/14-25-45.jpg with mode w: Permission denied
[1:ml1:Terras] [ERR] [ALL] put_picture: Can't write picture to file /media/usb-disk2/CCTV/Terras/2022-11-25/14-25-45.jpg - check access rights to target directory
Thread is going to finish due to this fatal error: Permission denied
[1:ml1:Terras] [ERR] [EVT] event_image_snapshot: Could not create symbolic link [2022-11-25/14-25-45.jpg]: Permission denied
`

zagrim, 2022-11-28:
You need to grant write access to the user MotionEye is running (which I think the default way of setting up is user "motion" which is member of group "motion"). The following commands should do the trick, I suppose:
`sudo chgrp -R motion /media/usb-disk2/CCTV
sudo chmod -R g+w /media/usb-disk2/CCTV`

That transfers group-level ownership of the CCTV-directory (and anything under it) to group "motion".

JamBalaya56562, 2026-11-10:
Closing, as zagrim answered this: since the Python 3 releases motionEye runs as the `motion` user, so the recording folder needs write access for it, e.g. `sudo chgrp -R motion <dir>` and `sudo chmod -R g+w <dir>`.

closed 2026-11-10 by JamBalaya56562