motioneye-project/motioneye #2657: Permission issues on a fresh install of Rasperry Pi OS Bullseye

roelvdwater, 2022-12-02:
First of all, thank you for the amazing work. I've been running motionEye for a while and as I got myself a new SD card to do a fresh install of Raspberry Pi OS, I decided to give the active development branch a try.

However, when I follow the setup instructions on this fresh install, I'm getting permission errors.

```
Dec 02 20:54:36 raspberrypi systemd[1]: Started motionEye Server.
Dec 02 20:54:43 raspberrypi meyectl[402]: configure_logging cmd motioneye: False
Dec 02 20:54:43 raspberrypi meyectl[402]: configure logging to file: None
Dec 02 20:54:43 raspberrypi meyectl[402]:     INFO: hello! this is motionEye server 0.43.0
Dec 02 20:54:43 raspberrypi meyectl[402]: CRITICAL: pid directory "/var/run" does not exist or is not writable
Dec 02 20:54:43 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 02 20:54:43 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
Dec 02 20:54:43 raspberrypi systemd[1]: motioneye.service: Consumed 5.831s CPU time.
```

I can (temporarily) fix the issue by granting permissions to all users to `/var/run`, but this temporary fix breaks when rebooting. I'm having the same permission issue when I'm trying to store recordings on a mounted USB drive. I didn't have these issues when I was running the "old" version of motionEye.

Please let me know if I have to provide more details.

MichaIng, 2022-12-02:
Thanks for reporting.

Did run `motioneye_init` to install the [shipped systemd unit](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/motioneye.systemd)? It contains `RuntimeDirectory=motioneye` which assures that the directory is created with correct permissions. Also assure that there is no override in `/etc/systemd/system/motioneye.service.d` which breaks this setting.

roelvdwater, 2022-12-08:
Sorry for the late reply. I wasn't fully sure if I ran it, so I ran it again to be sure but it has no effect, I'm still getting the error.
Not sure if this is expected, but the `/etc/systemd/system/motioneye.service.d` file doesn't exist on my machine. Could that be related?

ganzales, 2023-02-09:
Same issue after restoring config from an older version. Manual setup solved my problem.

roelvdwater, 2023-02-17:
> Same issue after restoring config from an older version. Manual setup solved my problem.

What do you mean with manual setup? I installed the latest (from a couple of months ago) version from the `dev` branch following the [Installation](https://github.com/motioneye-project/motioneye/tree/dev#installation) steps, which resulted in the issue described by me above.

ganzales, 2023-02-19:
> What do you mean with manual setup? I installed the latest (from a couple of months ago) version from the `dev` branch following the [Installation](https://github.com/motioneye-project/motioneye/tree/dev#installation) steps, which resulted in the issue described by me above.

If you restore config from backup, try add camera manually.

PhatHub, 2023-03-29:
I have instructions (in #2540 ) for setting up `tmpfiles.d` to create the correct /var/run/motion permissions so it will:

- Not run motioneye as root
- Re-creates the temporary (volatile)  file under /var/run with proper permissions upon reboot

MichaIng, 2023-03-29:
Is the PID file actually required? Probably disabling it is the simpler option?

PhatHub, 2023-03-29:
Traditionally it's better to have a service make the PID file, at least for easy detection of a process that's already running. I haven't dug into the code for it, there's a possibility that systemd would [rely on its existence](https://hexdocs.pm/distillery/config/pidfiles.html)?

Also, setting up the tmpfiles configuration is really easy, it's just a single line of text.

But that's up to the engineers of Motioneye to decide on.

MichaIng, 2023-03-30:
`systemd` does not need PID files as long as the process is running in foreground, which is true in case of  motionEye. If it needed it, one would just add `RuntimeDirectory=motioneye` to have `/run/motioneye` automatically created on service start with proper permissions. Actually this is what we are doing: https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/motioneye.systemd

I think we should consequently allow to have no PID file created and not create it by default. Currently this is not possible, but shouldn't be hard to implement.

But to solve the actual issue, we simply need to overwrite the systemd unit on motionEye installs/upgrades, as the new one has the required changes.

MichaIng, 2023-04-01:
@roelvdwater 
Btw, sorry for loosing track. Not the systemd unit was wrong, but the directory where the PID file was about to be created, as of the old config file. This fixes it:
```sh
sed -i '\|^run_path /var/run$|c\run_path /run/motioneye' /etc/motioneye/motioneye.conf
```
Both, the PID file setting and the systemd unit are now updated when installing the new motionEye version to assure old incompatible ones are not breaking startup: #2733

roelvdwater, 2023-10-05:
@MichaIng Sorry for the incredibly late reply but I haven't gotten to motionEye since today as I was still running on the old motionEye version and didn't bother upgrading until one of my SD cards broke down. It's working fine with the permissions now!