motioneye-project/motioneye #2661: Moviepasstrough - main stream/second stream

speeskeek, 2022-12-06:
hi. using motioneye since 6-7 years now . very happy. 
enabeled moevie passtrough yesterday. It reduced my load significantly. great option!

but - I'm using D1 resolution in motioneye (704x576 or something, secondary stream of the camera), but the camera's can also do 720p via another stream.
it would be great if motioneye can continue the D1 resolution on web interface (mainly watched on cell phone, so sont need big details) but the recordings can be at maximum resolution (main stream) to see details of all the nasty bulgars their faces :)

thanks

zagrim, 2022-12-08:
You can add [netcam_high_url](https://motion-project.github.io/motion_config.html#netcam_high_url) parameter to Extra Motion Parameters (the value should be the URL for the high-res stream), that should do what you want if I understood this correctly.

JamBalaya56562, 2026-11-10:
Closing, as zagrim answered this: Motion's `netcam_high_url` (`netcam_highres` up to Motion 4.3) under Extra Motion Options records the main stream with Movie Passthrough, while detection and the UI keep the substream.

closed 2026-11-10 by JamBalaya56562