motioneye-project/motioneye #2663: Working Schedule does not work (?)

Thormir84, 2022-12-09:
I've tried to set the working schedule for motion detection, but i think it doesn't work correctly.

I have set days and hour but the motion detection keeps sending notifications and recording videos.

What's wrong?

[image]

black00019, 2023-01-28:
Hi the issue is also open?
I have open a similar issue on Dietpi- MotionEye Github:
[https://github.com/MichaIng/DietPi/issues/6101](url)

The project is under development?

Thormir84, 2023-01-28:
The issue still exists for me, but i did not receive any feedback.

MichaIng, 2023-01-28:
Thanks for reporting. There is quite much stuff to do/review here, and this is not my main project. If anyone is able to read Python, HTML and in case Jinja2 templates, please test and have a look into it.

Are you guys seeing any errors either in browser console when applying/changing these settings or in motionEye logs?
```sh
journalctl -u motioneye
```

black00019, 2023-02-04:
I have this error:
`Feb 04 09:38:17 DietPi motion[1260512]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Feb 04 09:38:17 DietPi motion[1260512]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Feb 04 09:38:17 DietPi motion[1260512]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Feb 04 09:38:17 DietPi motion[1260512]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
Feb 04 09:38:17 DietPi motion[1260512]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Feb 04 09:38:17 DietPi motion[1260512]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Feb 04 09:38:20 DietPi motion[1260512]: [1:ml1:Studio] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
`

Thormir84, 2023-02-04:
I have no error.

hardwareadictos, 2023-02-06:
Same issue here, getting notifications whatever the working schedule is....

Erpiovra87, 2023-09-06:
Hi! any update? same problem here

ezequielitalia, 2024-05-04:
Hi!
If you need some help for testing please let me know... I've the same issue.
Running:
motionEye Version | 0.43.1b1
Motion Version | 4.6.0
OS Version | Linux 5.10.0-28-amd64

arseniov, 2024-05-21:
**+1**

Same issue here, I am running 0.43.1b1 on Docker :edge image and working schedule doesn't seem to be working at all. Whatever hour/schedule I set, motion detection is always on.

Marijn0, 2024-05-25:
Here is the output of my journalctl -u motioneye when I enable motion detection in the working schedule: 

May 25 18:38:03 raspberrypi motion[28489]: [0:cn0] [DBG] [STR] webu_parseurl: Sent url: /1/detection/status
May 25 18:38:03 raspberrypi motion[28489]: [0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /1/detection/status
May 25 18:38:03 raspberrypi motion[28489]: [0:cn0] [DBG] [STR] webu_parseurl: camid: >1< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2 :>< val2:><
May 25 18:38:03 raspberrypi motion[28489]: [0:wu0] [INF] [ALL] webu_answer_ctrl: Connection from: 127.0.0.1
May 25 18:38:03 raspberrypi meyectl[27974]: DEBUG: motion detection is disabled for camera with id <built-in function id>
May 25 18:38:03 raspberrypi meyectl[27974]: DEBUG: must enable motion detection for camera with id 1 (outside working schedule)
May 25 18:38:03 raspberrypi meyectl[27974]: DEBUG: enabling motion detection for camera with id 1

And this is what I get when I deactivate motion detection in the working schedule: 

May 25 18:36:13 raspberrypi motion[28231]: [0:cn0] [DBG] [STR] webu_parseurl: Sent url: /1/detection/status
May 25 18:36:13 raspberrypi motion[28231]: [0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /1/detection/status
May 25 18:36:13 raspberrypi motion[28231]: [0:cn0] [DBG] [STR] webu_parseurl: camid: >1< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2 :>< val2:><
May 25 18:36:13 raspberrypi motion[28231]: [0:wu0] [INF] [ALL] webu_answer_ctrl: Connection from: 127.0.0.1
May 25 18:36:13 raspberrypi meyectl[27974]: DEBUG: motion detection is disabled for camera with id <built-in function id>

The problem here (I think) is that motioneye thinks motion detection is disabled while this is not the case. 
When I run:

curl http://localhost:7999/1/detection/status
Camera 1 Detection status ACTIVE

This gives ACTIVE in capital letters and motioneye only searches for active,

url = f'http://127.0.0.1:{settings.MOTION_CONTROL_PORT}/{motion_camera_id}/detection/status'

     request = HTTPRequest(
         url,
         connect_timeout=_MOTION_CONTROL_TIMEOUT,
         request_timeout=_MOTION_CONTROL_TIMEOUT,
     )
     resp = await AsyncHTTPClient().fetch(request)
     if resp.error:
         return utils.GetMotionDetectionResult(None, error=utils.pretty_http_error(resp))

     resp_body = resp.body.decode('utf-8')
     enabled = bool(resp_body.count('active'))

When I edit: enabled = bool(resp_body.count('active')) to:
enabled = bool(resp_body.count('ACTIVE'))  
and when I deactivate motion detection I get this:

May 25 19:07:23 raspberrypi motion[30826]: [0:cn0] [DBG] [STR] webu_parseurl: Sent url: /1/detection/status
May 25 19:07:23 raspberrypi motion[30826]: [0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /1/detection/status
May 25 19:07:23 raspberrypi motion[30826]: [0:cn0] [DBG] [STR] webu_parseurl: camid: >1< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2 :>< val2:><
May 25 19:07:23 raspberrypi motion[30826]: [0:wu0] [INF] [ALL] webu_answer_ctrl: Connection from: 127.0.0.1
May 25 19:07:23 raspberrypi meyectl[30817]: DEBUG: motion detection is enabled for camera with id <built-in function id>
May 25 19:07:23 raspberrypi meyectl[30817]: DEBUG: must disable motion detection for camera with id 1 (during working schedule)
May 25 19:07:23 raspberrypi meyectl[30817]: DEBUG: disabling motion detection for camera with id 1

Problem solved I think......

zagrim, 2024-05-26:
Good catch! That would explain this well. Seems like `response.body.lower().count('active')` was accidentally changed to `resp_body.count('active')` (`.lower()` being dropped off) during refactoring here: https://github.com/motioneye-project/motioneye/commit/e1480197da90d7f035de419ff4b68d19f62bc153#diff-d3a426ec3ad26f1d6475f1bfe0d91afa364ad450736540a8487837405310dc6cL231-L238

Marijn0, 2024-05-26:
Thanks, hope someone can fix this in dev?

zagrim, 2024-05-27:
There's the PR, I hope someone can do a quick test with that.

MichaIng, 2024-05-28:
Merged, we'll push a new beta soon. I'll just test and in case merge #2903 as well.

PR #2987 (2024-05-28): Fix motion detection status check
This should fix motion detection check and make working schedule work as it did in the Python 2 releases.
Thanks for @Marijn0 for pointing out the flaw in the code.

I've not myself tested this, hopefully someone using working schedule can do that.

Fixes #2663
closed 2024-05-28 by MichaIng