motioneye-project/motioneye #2665: Frame Change Threshold 0.1% is not enough

Nuke142, 2022-12-12:
Motioneye does not start recording when really small amount of frames is changed (<1k).
is frame change threshold 0.1% not enough for?
[image]

zagrim, 2022-12-12:
It might be that if the camera resolution is large, 0.1% becomes large enough so that motion detection is not triggered for smallish changes. For that reason the percentage with a fixed amount of decimals together with linear scale might be too restricting. But to be able to set 0.05% (or even 0.01% - might well be realistic with 4K camera), 0.1%, 0.2% as well as, say, 18% without having to make the slider long enough that 1 pixel could represent 1/200th (slider minimum length of 200 pixels - or even 1/500th, this would lead to slider minimum length of 500 pixels) is a bit difficult given that the slider must fit in the relatively narrow side bar in the UI... :thinking: 
I think this is a case where logarithmic scale would be suitable to provide enough precision in the small end while reducing precision in the large end (where it is not likely that relevant if the set value is 18% or 19%. But logarithmic scales are non-intuitive for most people, which would make the slider behave weird when comparing to the rest of the sliders.

As for the specific case here, noise level of almost 9% seems quite much, unless you have to cope with low-light conditions. The view in the screenshot does not seem that noisy. Because too high noise level requires larger per-pixel value change in order for the change to be counting as motion, lowering noise level might be something you should try. Myself I have noise level set to 1% and do not have issues with noise triggering motion even though my cameras are outside pointing at very badly lit area (they do have IR leds, though, but those do not provide that much illumination in the overall scene).

zagrim, 2022-12-12:
> I still use old version on python2 because newest dev versions have a problems with videos thumbnails - their generating takes a whole life (1/per 30sec)

I don't think that has been reported so far... Please open another issue for this so that it can be tracked separately from this issue.

Nuke142, 2022-12-20:
> It might be that if the camera resolution is large, 0.1% becomes large enough so that motion detection is not triggered for smallish changes. For that reason the percentage with a fixed amount of decimals together with linear scale might be too restricting. But to be able to set 0.05% (or even 0.01% - might well be realistic with 4K camera), 0.1%, 0.2% as well as, say, 18% without having to make the slider long enough that 1 pixel could represent 1/200th (slider minimum length of 200 pixels - or even 1/500th, this would lead to slider minimum length of 500 pixels) is a bit difficult given that the slider must fit in the relatively narrow side bar in the UI... 🤔 I think this is a case where logarithmic scale would be suitable to provide enough precision in the small end while reducing precision in the large end (where it is not likely that relevant if the set value is 18% or 19%. But logarithmic scales are non-intuitive for most people, which would make the slider behave weird when comparing to the rest of the sliders.
> 
> As for the specific case here, noise level of almost 9% seems quite much, unless you have to cope with low-light conditions. The view in the screenshot does not seem that noisy. Because too high noise level requires larger per-pixel value change in order for the change to be counting as motion, lowering noise level might be something you should try. Myself I have noise level set to 1% and do not have issues with noise triggering motion even though my cameras are outside pointing at very badly lit area (they do have IR leds, though, but those do not provide that much illumination in the overall scene).

Much better after i set noise level 3% and 720p instead of 1080p - changed  both in camera native settings and in motioneye. But not ideal, Far objects are still not captured. Still need to catch smaller pixels. I can't imagine how bad it works if i set 4k camera.
[image]

zagrim, 2022-12-20:
Yes, I think 4K cameras are very good reason to try to figure out a solution for setting low enough frame change threshold.

As a possible workaround for the limitation, in case you haven't tried it yet, enabling "Despecle Filter" might improve detection if the whole figure of the person moving isn't detected as moving (since it makes groups of nearby changed pixels combine, thus increasing the area that is detected to be changing).

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #697, which asks for finer frame change thresholds than 0.1%.

closed 2026-11-10 by JamBalaya56562