motioneye-project/motioneye #2672: No recordings

Ryan5374, 2022-12-21:
I installed motioneye using the ubuntu tutorial on a linux mint system. Everything works fine, motion detection even shows the red boxes when movements are made, but there are never any videos recorded. I have file storage to a local directory, and motion detected recording triggers. I have the mint-meta package installed for codecs and I've tried multiple formats, including MKV and H264 MP4, but I never get any videos or photos, even when clicking the manual photo button. 
[image]

zagrim, 2022-12-25:
Kind of obvious, but have to ask since you provide next to no info of your configuration: Is storing movies enabled? Is the recording mode Motion Triggered? Do the logs show any errors?

Ryan5374, 2022-12-27:
My bad, I'll try to include that info next time. I updated my codecs and that ended up working, sorry for bothering you

closed 2022-12-27 by Ryan5374