motioneye-project/motioneye #2673: Unable to open video device

fallingupdownthere, 2022-12-22:
I have setup an RTSP camera on a Pi 3 using an Arducam camera and installed MotionEye in HomeAssistant.  The feed worked fine but after a couple of hours of tinkering with settings I received a "Unable to open video device" message in the feed window.  I checked the stream URL in VLC and it worked fine.  After rebooting the Pi and HomeAssistant I was still receiving the error so I deleted the cam and added it back in.  That seemed to fix the issue and I kept tinkering with the settings.  Mainly motion detection, framerates, and notifications.  I got the webhook working to notify me via HomeAssistant on motion but then the stream failed again and here I sit.  Any suggestions?  Thanks

fallingupdownthere, 2022-12-22:
After some more fiddlng I have noticed that even though it won't display a feed it will still detect motion and record on motion.  Also, the feed was working in the HA app while not on my desktop but now both feeds are down.

JamBalaya56562, 2026-11-11:
Closing, as this got no replies. That text is shown while Motion cannot read the camera's stream, even when detection and recording work in between, and `motion.log` says why. Please open a new issue with that log if it still happens with a current add-on.

closed 2026-11-11 by JamBalaya56562