motioneye-project/motioneye #2677: No filesystem created during installation

mattpstanding, 2022-12-29:
Hi,

I am trying to install MotionEye on a Pi3B running Pi OS 64bit Lite (Bullseye). I have followed the installation guide here:

[https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye) 

All the commands complete successfully with no errors. However when I try and start the service and review the log it states that there is no MotionEye.config file. When I review the filesystem there is no motioneye folder.

I have tried this three times now with the same results. Is there an issue?

Matt

rasmuswork, 2023-03-02:
I have the same issue and have been unable to find a workaround. I've tried reinstalling both Raspbian bullseye and the program.
I've tried multiple different installation guides after the Raspbian-Bullseye did not work.
I've tried on a freshly installed version of Raspberry-Bullseye while strictly following the instructions, yet I've still been unable to get it up and running, due to no motioneye folder being created.

My native language is not english; please excuse any typing errors.

MichaIng, 2023-03-02:
Please follow the instructions here: https://github.com/motioneye-project/motioneye/tree/dev#installation

JamBalaya56562, 2026-11-12:
The wiki guide for Bullseye installed the old Python 2 release. With the steps in the README (https://github.com/motioneye-project/motioneye#installation), which MichaIng linked, `sudo /opt/motioneye/bin/motioneye_init` creates `/etc/motioneye/motioneye.conf` and the systemd service. Closing as answered; please open a new issue if the current steps fail for you.

closed 2026-11-12 by JamBalaya56562