motioneye-project/motioneye #2678: Cant deactivate Authentication for Streaming

Floatcher, 2022-12-30:
Hi,
I deactivated Authentication, but it is still promted on my tablet. Sadly the keyboard is not working on my tablet with the embeded stream so I cant add any credentials and use the rember function. How can I disable the login for the streaming?
Thx Flo

[image]

[image]

moition.conf
# @admin_username admin
# @admin_password 
# @normal_username user
# @normal_password 
# @lang en
# @enabled on

setup_mode off
webcontrol_port 7999
webcontrol_interface 1
webcontrol_localhost on
webcontrol_parms 2
camera camera-1.conf

Floatcher, 2022-12-30:
I made so more tests. The issue only exists for the embedded stream. For the normal "Streaming URL" the configuration is working and deactivates the login. Please add this also to the "Embed URL"

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #2686: as @MichaIng explained there, the embedded frame needs a motionEye session, so it asks to log in on purpose. The *Streaming URL*, which follows the stream authentication setting, works without a login, as found above.

closed 2026-11-13 by JamBalaya56562