motioneye-project/motioneye #2679: Grey screen for RLC-520 camera

kevinprovoost, 2022-12-30:
Hi, I'm having troubles getting a steady camera feed with my RLC-520:

The link i'm using:
rtmp://192.168.0.24:1935/bcs/channel0_main.bcs?channel=0&stream=0&user=user&password=***

It is showing the FPS rate but only a grey image.
In VLC this works fine.

The rtsp link:
rtsp://user:***@192.168.0.24:554/h264Preview_01_main
Gives me a fine session at first but after a couple if minutes, it quickly turns into weird images.
It is run on a desktop pc so cpu usage is not a problem, I have 3 other dlink cameras on motioneye with rtsp without blurry or weird images.

kevinprovoost, 2022-12-30:
Must have had a broken config.
Reinstalled motioneye with the latest docker config, added rmtp link and now it's running fine.

Phoenix1747, 2023-01-11:
Then please close the issue.

closed 2023-01-11 by kevinprovoost