motioneye-project/motioneye #2683: RPi Camera 3 compatibility

Firesphere, 2023-01-09:
Will MotionEye, in its current state, be compatible with the new RPi cam 3?
https://www.raspberrypi.com/news/new-autofocus-camera-modules/

> The new Camera Module 3 is only supported by the modern [libcamera](https://www.raspberrypi.com/documentation/computers/camera_software.html#libcamera-and-libcamera-apps) software environment and by the libcamera-based [Picamera2](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) beta under Raspberry Pi OS Bullseye, and not by the [legacy closed-source camera stack](https://www.raspberrypi.com/documentation/computers/camera_software.html#raspicam-applications)

hotducker, 2023-01-21:
I just try RPI3b (MotionEyeOS) + RPI Camera 3, It's not work and still find the solution.

gpambrozio, 2023-01-21:
I found a way to make it work with the current motioneye version on a pi zero w with the new camera.

First, create a file named `motion.sh` in `/home/pi`:

```
#!/bin/bash

/usr/bin/libcamerify /usr/bin/motion $@
```

Now, edit your `motioneye.sh` file (should be in `/etc/motioneye`) and make sure that the line with `motion_binary` looks like this:

```
motion_binary /home/pi/motion.sh
``` 

Now you can configure your camera as a local camera on `/dev/video0`

The information that led me to this solution was found [here](https://motion-project.github.io/motion_config.html#basic_setup_picam)

MidnightLink, 2023-01-22:
> ```
> motion_binary /home/pi/motion.sh
> ```

I tried this but unfortunately it doesn't work. A couple thingsI've noticed, though.

 `motioneye.sh` is actually `motioneye.conf` and the correct chmod permissions need to be given to the `motion.sh` 

 `libcamera-tools` has to be installed

I edited my `camera-1.conf` and changed the video device to `/dev/video0` and when running `libcamera-hello` I'm getting a message that the device is in use...so it feels like it's close!

gpambrozio, 2023-01-22:
@MidnightLink I edited my post to current the file and add the `libcamera-tools` (I didn't need that as I have the latest debian and it already includes it).

I think your issue is that you can't be running `libcamera-hello` (or any other application that tries to use the camera) when you start motioneye

johnm-w, 2023-01-26:
> Now restart motioneye and you can configure your camera as a local camera on `/dev/video0`

Hi @gpambrozio - can you explain how to do this?  In MotionEye, Add Camera, I select "Local V4L2 Camera" from the "Camera Type" dropdown, then it lists several options for "Camera".  With the module 2 camera, I had to try each one until I found the one that worked, but I can't manually enter a device path.

Also, please can you let me know the exact OS you're running?  I tried your fix on Buster & Bullseye with no luck (on an RPi4).  

Thanks!

gpambrozio, 2023-01-27:
> Hi @gpambrozio - can you explain how to do this? In MotionEye, Add Camera, I select "Local V4L2 Camera" from the "Camera Type" dropdown, then it lists several options for "Camera". With the module 2 [...]

Best way might be to just pick any cameras, then shutdown motioneye and edit `camera-1.conf` (probably in `/etc/motioneye`) and make sure that the `videodevice` line is `videodevice /dev/video0`. The start motioneye again.

> Also, please can you let me know the exact OS you're running? I tried your fix on Buster & Bullseye with no luck (on an RPi4).

Using bullseye on a pi zero w.

MisterEmm, 2023-01-27:
Tried this earlier on a Pi 3B with the latest Buster release, fresh install of motioneye, followed the steps above and it works well, I can see the view from the camera via the web interface. For now it doesn't seem to want to let me make any changes, keeps saying "Refresh Recommended" if I try to change resolution or something. I can see the changes reflected in camera-1.conf but the display doesn't change. Will have another poke at it tomorrow, but grateful to have got this far thanks to the previous posts.

johnm-w, 2023-01-27:
Thanks @gpambrozio - I have it working to a point now - I can see the camera's feed.  Unfortunately:
I don't see all the camera's controls in the Video Device section (like I do with the camera module 2) and performance isn't great.

And like @MisterEmm , If I make any changes via the web app: "An error occurred. Refreshing is recommended." - but refreshing does appear to persist settings

 I've also had very similar results running:

libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream1}' :demux=h264

(https://www.raspberrypi.com/documentation/computers/camera_software.html#libcamera-vid)

Then adding the camera as a network camera in motioneye using:

rtsp://localhost:8554/stream1

then selecting the UDP option that appears.

Neither option seems as quite as good as when I'm using the v2 camera - but maybe I just need to experiment/tweak more.  Or hopefully someone with the right skills can update motioneye.

Also, @gpambrozio please could you explain what your motion.sh script does?

gpambrozio, 2023-01-27:
@johnm-w `motion.sh` is a wrapper to make motion use `libcamerify` as described in [this document](https://motion-project.github.io/motion_config.html#basic_setup_picam). Without that `/dev/video0` doesn't work as motion expects it.

I tried just changing `motion_binary`  to `libcamerify motion` but that doesn't work so I created this simple script to trick motioneye.

ReneMuetti, 2023-01-28:
I have proceeded according to these instructions. The camera can now be integrated into MotionEye and a picture is displayed.
https://github.com/motioneye-project/motioneye/issues/2683#issuecomment-1399367885
but unfortunately errors are still written to the log file:
`[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[2:22:53.594696843] [21329]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[2:22:53.617144741] [21330] ERROR V4L2 v4l2_device.cpp:352 'imx708_wide': Unable to set controls: Das Gerät oder die Ressource ist belegt
[2:22:53.619075612] [21330]  WARN CameraSensorProperties camera_sensor_properties.cpp:205 No static properties available for 'imx708_wide'
[2:22:53.619170872] [21330]  WARN CameraSensorProperties camera_sensor_properties.cpp:207 Please consider updating the camera sensor properties database
[2:22:53.665191035] [21330]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[2:22:53.665626136] [21329]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
[2:22:53.665809312] [21329] ERROR V4L2Compat v4l2_camera.cpp:34 Failed to acquire camera
/home/pi/motion.sh: Zeile 3: 21328 Speicherzugriffsfehler  /usr/bin/libcamerify /usr/bin/motion $@
`

Waldexxx, 2023-01-30:
Is there a way to implement it in MotionEyeOS?

mgrn0, 2023-02-11:
unfortunately I could not get my camera model 3 to work via any of the suggestions above with a raspberry pi 4 / b running bullseye. Really hope the new cameras will be supported without workarounds in motioneye soon.

MaRaGaWa, 2023-03-14:
@gpambrozio 
It works ! Many thanks !

My setup :
- RPI 4 with 1 GB RAM, 16 GB SD-Card
- RPI OS, Rasbian GNU/Linux 11 (Bullseye)
- Motioneye vs 0.42.1,  Motion vs 4.3.2
- Raspberry Pi Camera Module 3 NoIR

Some additional advices/observations:
- after adding Camera1 in Motioneye make sure you select the upper 'Unicam'
- note that the 'camera-1.conf' file does not pop-up in /etc/motioneye than after adding that Camera1,
- then edit the 'camera-1.conf' as in the above instructions 
- and... you can additionally improve the picture-quality dramatically by editing 'stream_quality' and 'movie_quality' in 'camera-1.conf'(I both increased them to 150, perfect quality !)
- And the infrared also works, and even automatically :-)

notabotscott, 2023-03-22:
For those who have gotten this workaround to work:
- Are you using a 64 bit or 32 bit OS version of Bullseye?
- Did you enable legacy camera support in raspi-config?
- Are you adding the camera as V4L2 or MMAL?

I just keep getting the following errors no matter what i try:

```
[1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Not a V4L2 device?
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
```

B4n4n3, 2023-03-26:
Hey together, what am I doing wrong?

- Raspberry Pi 3 B+
- Bullseye OS 32-bit lite
- installing motioneye according to the official [instructions](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye)
- follow the instructions from @gpambrozio 
    - creating motion.sh an setting up chmod (sudo chmod +x motion.sh)
    - editing `motioneye.conf` by following the instructions.
    - start  motioneye and added a V4L2 Camera
    - stop motioneye an edit `camera-1.conf`  -> `videodevice /dev/video0`
    - start motioneye ... 
 
but no picture in the preview.

notabotscott, 2023-03-27:
> Hey together, what am I doing wrong?
>
> * Raspberry Pi 3 B+
> * Bullseye OS 32-bit lite
> * installing motioneye according to the official [...]

Did you install the motioneye dev branch? https://github.com/motioneye-project/motioneye/tree/dev
Also, did you install libcamera-tools? `sudo apt install libcamera-tools`

B4n4n3, 2023-03-31:
> Did you install the motioneye dev branch? https://github.com/motioneye-project/motioneye/tree/dev Also, did you install libcamera-tools? `sudo apt install libcamera-tools`

Hey @notabotscott 

thank you very much, the wrong installation was the problem.
I had already installed the libcamera tools.
But i had to install the motioneye dev branch.

BobSmithPi, 2023-05-07:
Thank you for this workaround.  I was able to get my camera module 3 working fine except for adding parameters in the config.  Is there any way to change brightness, exposure and lens-position? I've tried adding them different ways in the config file and nothing seems to work.  Any help would be greatly appreciated.

bigOconstant, 2023-05-15:
None of these worked for me on a rpi4 running latest 64 bit os. Really wish I could use this new camera.

m4rky-m4rk, 2023-05-16:
I have few cameras lying idle because of the issue and lack of time to dig into the issues.

strips, 2023-05-16:
I never got the libcamerify to work but still feel it should have. Might give it a new try one day. Anyways the motion project is old and will never support the new cameras using libcamera. MotionEye should maybe start looking at transitioning to motionPlus (when python3 has reached maturity) as it seems a replacement wich is actively developed. motionPlus works with libcamera.

I found this solution online and you need screen and ngnix installed. I created a shell script to start by boot in crontab. This has been stable now for some months. This is the first time I ever used ngnix. The two processes use maybe 5% CPU in average. It can support a good FPS but I keep it low for motion detection. 

`# cat /root/start_rpicam3_stream.sh`
```
#!/bin/bash -x

mkfifo /root/stream
chmod 777 /root/stream

screen -l -dmS libcamera /usr/bin/libcamera-vid -t 0 --width 1920 --height 1080 --codec h264 -n --hdr 1 --profile=high --framerate 5 -o /root/stream
screen -l -dmS ffmpeg /usr/bin/ffmpeg -y -i /root/stream -s 1920x1080 -r 5 -c:v copy -b:v 1500k -maxrate 5000k -bufsize 10000k -g 60 -f flv rtmp://localhost/live/stream
```

`# crontab -l`
```
# Edit this file to introduce tasks to be run by cron.
# m h  dom mon dow   command
@reboot sleep 60 && /root/start_rpicam3_stream.sh
```

`# cat /etc/nginx/nginx.conf`
```
rtmp {
        server {
                listen 1935;
                allow publish 127.0.0.1;
                deny publish all;
                timeout 60s;
                notify_method post;
                chunk_size 8192;

                application live {
                        live on;
                        record off;
                }
        }
}
```

Camera device/URL
`netcam_url rtmp://localhost/live/stream`

Change the FPS to what you want. 1920x1080 is maximum resolution for the RPI hw codec as I think it's used here.

mgrn0, 2023-05-16:
@strips are you using this as a replacement for motioneye or just as a streaming source / camera FOR motioneye?

strips, 2023-05-16:
@mgrn0, I'm using this with motionEye. Effectively creating a network stream from the camera and configuring motionEye to use the stream as a source.

bigOconstant, 2023-05-16:
I'm kinda confused on what you are using screen for here and why it's needed. But I like the concept of connecting to it as a network stream.  That's a great idea I'll play with this this weekend.

strips, 2023-05-16:
@bigOconstant screen is a simple (ugly) way to run a background process. Could probably use nohup or set up proper daemon instead. I tried nohup but one of the commands did not like the limited environment and went back to screen.

Dquaglio, 2023-05-16:
Is there any way to achieve these hotfixes on MotionEye OS directly? not installed in Raspberry Pi OS

strips, 2023-05-19:
I got libcamerify to partially work by compiling Motion from source. My next issue was that MotionEye did not seem to detect the running process of Motion when using the wrapper and got a lot of crash loops. So close, I decided to see if I could fix this in the code to eliminate the need for an external wrapper script. In my own fork I have implemented support for libcamerify and it can be configured in motioneye.conf. So far it seems to work pretty good. The only issue seems to be setting parameters for the camera. I'll probably create a PR for my fix. Anyways the code is here: https://github.com/strips/motioneye/tree/feature/libcamerify

github-0, 2023-05-20:
@strips Thanks for the instructions for libcamera+ffmpeg combo as I couldn't get the libcamerify method to work with camera module 3 (1.1 worked fine) and what's more, with libcamerify there is no way(?) to set camera parameters which is a must in my use case. 

I had to add -re parameter to _ffmpeg -y -re -i /root/stream -s 1920x1080 -r 5 -c:v copy -b:v 1500k -maxrate 5000k -bufsize 10000k -g 60 -f flv rtmp://localhost/live/stream_ as otherwise ffmpeg was not reading the input in real time, reached end of file and quit.

One question about the named pipe (fifo) though - it seems this file (/root/stream) keeps increasing in size. How did you solve this?

strips, 2023-05-21:
@github-0 np!
I found the link where I found the workaround: https://www.wetransco.de/debian-bullseye-libcamera-and-motion-on-a-raspberry-pi-with-the-new-camera-module-v3-a-great-workaround/

Using [rtsp-simple-server](https://github.com/aler9/rtsp-simple-server/) should/could be a much simpler approach. I tested this for an hour before switching to libcarmerify and it worked at least as well. The advantage is that it's only one binary/command to run. If needed I could post the config file I made.

bigOconstant, 2023-05-21:
Thanks @strips I got it to work with your script in the same fashion and your 60 second sleep make it work after a minute on restart which is great. 

Full instructions for others would involve, starting from blank rasbian os, installing nxinx, follow the install motion-eye instructions from dev branch,  adding the nginx config file and then doing whats in the above comment and it works.

RippinAndTerran, 2023-06-10:
Hey @strips if you had the chance to upload your config file for rtsp-simple-server that would be amazing! I am pretty new at all of this and after a few days of trying things it is hard to tell where I need to dedicate my efforts to get it up and running. I am about to go down the ngnix road but since rtsp-simple-server is supposed to be easier I am a little apprehensive since I can't even get that to work.

techd4ve, 2023-06-14:
Hi Guys! Thanks for the workaround, it worked really well.
I also have the "apply" error bug but thing could be worse.

Do you may know how I can adjust the focus of the Cam module 3?
My Problem is that something is in front of the cam (left side ca. 25% of the image and top 40% of the image) and because of that, the Autofocus is not adjusted correctly. Is there any change I can set up a certain point for the autofocus or set it to manuell focus on that certain point?

I tried something with the tuning file but I didn't seem to work (maybe I also did it wrong)

Anybody an Idea how to fix it?

RenatoSerra22, 2023-06-30:
thanks for the solution @strips with this approach does the HD picture on motion function work or is it just the video stream?

strips, 2023-07-03:
> rtsp-simple-server

Hi, I put the file on github below. I haven't used MediaMTX since I tested it but I'lll probably go back to it. I'm having issues with libcamerify and keep having to restart the daemon. 
https://github.com/strips/stuff/blob/c8b10f50fa4aba3a685ad7890710de9c15716eb6/hakkebo_rpi/MotionEyeConfig/mediamtx/mediamtx.yml

strips, 2023-07-03:
@RenatoSerra22 

> thanks for the solution @strips with this approach does the HD picture on motion function work or is it just the video stream?

Which of the 3 solutions are you referring to :-)

1. Workaround with ffmpeg and nginx
Works pretty well but a lot of moving parts. HD picture works in stream and motion. But when triggering a motion it slows down and lags alot.

3. Workaround with rtsp-simple-server / MediaMTX
Works pretty well and replaces both ffmpeg and nginx processes in the first workaround. HD picture works in stream and motion. But when triggering a motion it slows down and lags alot. The added benefit MediaMTX has a lot og options to control the camera. Seems they have exposed most of the settings in libcamera API. Still I have not testes many of the options.

4. Libcamerify compatabilitylayer
Rather unstable. I keep having to restart the motionEye daemon. Not really sure but it might be enough to just restart the motion process. It might be my motionEye patch that does not work properly and motionEye is not able to detect the issue and restart motion. 

Personally I will go back to number 3.

kendoka15, 2023-07-06:
@strips Thanks for the ffmpeg+nginx method, works well. Libcamerify didn't work for me as from what I gather I couldn't have access to all the libcamera settings. I haven't tried MediaMTX yet as I needed a solution pretty quickly but if everything is controllable including shutter speed and gain I'll be trying it soon.

I ended up feeding the stream from the camera Pi Zero W to a Pi 4 I'm also using for Home Assistant and Samba as the Pi Zero 2 W I was previously using with MotionEye OS couldn't run MotionEye at 1080p 10fps without overheating (the camera has to work well in 40C weather). MotionEye OS previously worked fine in those conditions at 720p but either the resolution bump is too much or there was some hardware accelerated special sauce in MotionEye OS that isn't there in MotionEye

Anyway, thanks

D-an-W, 2023-09-22:
> I found a way to make it work with the current motioneye version on a pi zero w with the new camera module 3.
> 
> First, create a file named `motion.sh` in `/home/pi`:
> 
> ```
> #!/bin/bash
> 
> /usr/bin/libcamerify /usr/bin/motion $@
> ```
> 
> Now, edit your `motioneye.conf` file (should be in `/etc/motioneye`) and make sure that the line with `motion_binary` looks like this:
> 
> ```
> motion_binary /home/pi/motion.sh
> ```
> 
> Depending on your debian distribution you might have to also install `libcamera-tools` with `sudo apt-get install libcamera-tools`
> 
> Now restart motioneye and you can configure your camera as a local camera on `/dev/video0`
> 
> The information that led me to this solution was found [here](https://motion-project.github.io/motion_config.html#basic_setup_picam)

Got it working thanks, I had to change the permissions of the `motion.sh` file before it would show the camera however.

I used 755, not sure if thats the best however?

Wolfbeta, 2023-10-07:
> I found a way to make it work with the current motioneye version on a pi zero w with the new camera module 3.
>
> First, create a file named `motion.sh` in `/home/pi`:
>
> ```
> #!/bin/bash [...]

This worked for me on a RPi 4B + Bullseye 32-bit (64-bit did not work).

Stepyon, 2023-10-29:
> Hi Guys! Thanks for the workaround, it worked really well. I also have the "apply" error bug but thing could be worse.
>
> Do you may know how I can adjust the focus of the Cam module 3? My Problem is [...]

Autofocus doesn't work for me at all with "libcamerify" script workaround. Have you found how to make it working?

Ghosti10, 2023-11-09:
**Ok guys , the trick is super easy.**  
The Module3 Cam has issues with motioneye. 
but it is working perfcet with Spyglass.  Here is the trick: Install spyglass, start run.py -r 1920x1080  which is creating a stream. (http://youIP:8080/stream)  check this out for installation: https://robertmarks.cloud/posts/2023/02/setting-up-pi-camera-module-3-in-mainsailos/    (credit to Rob Marks)

then install Motioneye:  https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye

then DO NOT ADD MJPEG Cam in motioneye. Add a NETWORK Camera to motioneye,
and you have a perfect working motioneye with motion detection .

pingufreak, 2023-11-18:
Much easier:

https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-1817391735

clds84, 2023-11-23:
> I found a way to make it work with the current motioneye version on a pi zero w with the new camera module 3.
>
> First, create a file named `motion.sh` in `/home/pi`:
>
> ```
> #!/bin/bash [...]

Trying this now, but I don't have a 'pi' directory following /home to create a motion.sh file. Referencing other pi projects, pi would be a user, right?

pingufreak, 2023-11-23:
Hi,

just change the systemd config as described.

Kr

Pingufreak

Juanderer98, 2023-11-28:
> **Ok guys , the trick is super easy.** The Module3 Cam has issues with motioneye. but it is working perfcet with Spyglass. Here is the trick: Install spyglass, start run.py -r 800x600 which is creating a stream. (http://127.0.0.1:8080/stream) check this out for installation: https://robertmarks.cloud/posts/2023/02/setting-up-pi-camera-module-3-in-mainsailos/ (credits to Rob Marks)
> 
> then install Motioneye: https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye
> 
> then DO NOT ADD MJPEG Cam in motioneye. Add a NETWORK Camera in motioneye, and you have a perfect working motioneye with motion detection . And YES, autofocus is working as well. Enjoy

********************
I gave this a try but had problems toward the end.  I had set up a raspian Bullseye 64bit system on a RPi4b 4Gb machine and was able to view the RPi camera v3 preview but in following the instructions to install motioneye on Bullseye I wound up installing python 2 pip. Motioneye worked but then spyglass would no longer work. I tried entering:

~/spyglass $ ./run.py -r 1920x1080

in a terminal window, which had worked before the motioneye install, but now it gets an error from an incorrect version (I think) of python. Is there a way to direct the machine to use python3 for spyglass?

strips, 2023-11-28:
> I gave this a try but had problems toward the end. I had set up a raspian Bullseye 64bit system on a RPi4b 4Gb machine and was able to view the RPi camera v3 preview but in following the [...]

Did you try Motioneye dev branch? It's based on Python 3. Not sure with the builds but I think they are based on the default (old) python2 version.

Peuz85, 2023-11-30:
**RaspiOs bookworm 64 bit with raspicam v3 (finally works)**

`sudo apt update && sudo apt upgrade -y`

`sudo rm -rf /usr/lib/python3.11/EXTERNALLY-MANAGED`

`sudo apt --no-install-recommends install ca-certificates curl python3 python3-distutils python3-dev libcurl4-openssl-dev gcc libssl-dev`

`curl -sSfO 'https://bootstrap.pypa.io/get-pip.py'`

`sudo python3 get-pip.py`

`sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'`

`sudo apt install -y libcamera-v4l2`

`sudo apt install -y libcamera-tools`

`sudo motioneye_init`

`sudo sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service
`

`sudo systemctl daemon-reload`

`sudo systemctl enable motioneye --now`

`sudo reboot`

http://yourpiaddress:8765/

It can happen that when changes are made in the MotionEye, the message “unable to connect camera” appears.
`sudo reboot` (takes a little longer) and then it works

Have fun ;-)

D-an-W, 2023-12-03:
@Peuz85, just trying your method thanks!

I see the message below, is that normal?

Also, what type of camera do I select please?

```
pi@MotionEye:~ $ sudo sed -i 's//usr/local/bin/meyectl//usr/bin/libcamerify /usr/local/bin/meyectl/' /etc/systemd/system/motioneye.service
sed: -e expression #1, char 8: unknown option to `s'
```

Peuz85, 2023-12-03:
I select the first cam in the drop down menu and it works.

I don't see that message... I just installed Motioneye again and didn't get any errors

D-an-W, 2023-12-03:
> I select the first cam in the drop down menu and it works.
> 
> I don't see that message... I just installed Motioneye again and didn't get any errors

Any idea what the error means or how to fix it as the camera doesn't currently work following your guide to the letter?

D-an-W, 2023-12-03:
> He copied just my answer... You need to escape the slashes.
> […](#)

All working now, thank you!

Peuz85, 2023-12-03:
> He copied just my answer... You need to escape the slashes.

Du hast leider keine schritt für schritt anleitung geschrieben. Mit deiner "anleitung" so wie du sie geschrieben hast, ging leider gar nichts. Zudem kommt bei bookworm immer der fehler mit "... externally Environment" und dies muss man auch erst mal umgehen.
Danke für deine vorarbeit.

Lg und Grüsse aus Ö  ;-)

Stepyon, 2023-12-04:
> **RaspiOs bookworm 64 bit with raspicam v3 (finally works)**
> 
> > sudo apt update && sudo apt upgrade -y
> 
> > sudo rm -rf /usr/lib/python3.11/EXTERNALLY-MANAGED
> 
> > sudo apt --no-install-recommends install ca-certificates curl python3 python3-distutils python3-dev libcurl4-openssl-dev gcc libssl-dev
> 
> > curl -sSfO 'https://bootstrap.pypa.io/get-pip.py'
> 
> > sudo python3 get-pip.py
> 
> > sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
> 
> > sudo apt install -y libcamera-v4l2
> 
> > sudo apt install -y libcamera-tools
> 
> > sudo motioneye_init
> 
> > sudo sed -i 's//usr/local/bin/meyectl//usr/bin/libcamerify /usr/local/bin/meyectl/' /etc/systemd/system/motioneye.service
> 
> > sudo systemctl daemon-reload
> 
> > sudo systemctl enable motioneye --now
> 
> > sudo reboot
> 
> http://yourpiaddress:8765/
> 
> It can happen that when changes are made in the MotionEye, the message “unable to open video device” appears. `sudo reboot` (takes a little longer) and then it works
> 
> Have fun ;-)

This solution works except autofocus, is any way to make the autofocus working?

axeljerabek, 2023-12-07:
I also added a libcamera/rpicam-vid version in another article:
https://www.wetransco.de/debian-bookworm-and-libcamera-module-3/

Stepyon, 2023-12-08:
> This solution works except autofocus, is any way to make the autofocus working?

I was able to control the focus manually from the camera window by adding the action buttons:
[image]

1) Create a **zoom_in_1** script file (where "1" is camera number) in /etc/motioneye/ folder:
**sudo nano /etc/motioneye/zoom_in_1**

```
#!/bin/bash

# Define the device
device="/dev/v4l-subdev1"

# Define the maximum focus value
max_focus=960

# Get the current focus value
current_focus=$(v4l2-ctl -d $device --get-ctrl=focus_absolute | awk '{print $2}>

# Check if current focus is a number
if ! [[ $current_focus =~ ^[0-9]+$ ]] ; then
   echo "Error: Failed to get current focus value" >&2; exit 1
fi

# Increment the focus value by 20
new_focus=$((current_focus + 20))

# Check if the new focus exceeds the maximum
if [ $new_focus -gt $max_focus ]; then
    new_focus=$max_focus
fi

# Set the new focus value
v4l2-ctl -d $device -c focus_absolute=$new_focus
```
Create **zoom_out_1** script in the same folder:
**sudo nano /etc/motioneye/zoom_out_1**

```
#!/bin/bash

# Define the device
device="/dev/v4l-subdev1"

# Define the minimum focus value
min_focus=320

# Get the current focus value
current_focus=$(v4l2-ctl -d $device --get-ctrl=focus_absolute | awk '{print $2}>

# Check if current focus is a number
if ! [[ $current_focus =~ ^[0-9]+$ ]] ; then
   echo "Error: Failed to get current focus value" >&2; exit 1
fi

# Decrease the focus value by 20
new_focus=$((current_focus - 20))

# Check if the new focus is below the minimum
if [ $new_focus -lt $min_focus ]; then
    new_focus=$min_focus
fi

# Set the new focus value
v4l2-ctl -d $device -c focus_absolute=$new_focus
```
Make them executable:
**sudo chmod +x zoom_in_1
sudo chmod +x zoom_out_1**

2) Autofocus even when it is not working will reset the focus value every time you are trying to change it, to fix that you need to modify your camera json file which located here: /usr/share/libcamera/ipa/rpi/vc4 (in my case it is imx708_noir.json). Remove or just rename "rpi.af" section. See more details [here](https://forums.raspberrypi.com/viewtopic.php?t=353887)

Now you should be able to control the camera focus manually straight from the camera window.

haklan, 2023-12-18:
@Peuz85  I installed motion on Bookworm according to your instruction snd trying to setup a RPI HQ camera but I got "UNABLE TO OPEN VIDEODEVICE" in the stream. Reboot does not help. Got  "VIDIOC_STREAMON: Invalid argument" in systemctl status. Do you have any ideas?

● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; preset: enabled)
     Active: active (running) since Mon 2023-12-18 15:44:51 CET; 8min ago
   Main PID: 561 (meyectl)
      Tasks: 10 (limit: 765)
        CPU: 10min 22.668s
     CGroup: /system.slice/motioneye.service
             ├─561 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             ├─734 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             └─765 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5

Dec 18 15:52:40 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] v4l2_mmap_set: Error starting stream. VIDIOC_STREAMON: Invalid argument
Dec 18 15:52:40 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Dec 18 15:52:50 birdcam motion[765]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Dec 18 15:52:50 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] v4l2_fps_set: Error setting fps. Return code -1
Dec 18 15:52:50 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] v4l2_mmap_set: Error starting stream. VIDIOC_STREAMON: Invalid argument
Dec 18 15:52:50 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Dec 18 15:53:00 birdcam motion[765]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Dec 18 15:53:00 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] v4l2_fps_set: Error setting fps. Return code -1
Dec 18 15:53:00 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] v4l2_mmap_set: Error starting stream. VIDIOC_STREAMON: Invalid argument
Dec 18 15:53:00 birdcam motion[765]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open

JustGlowing, 2024-01-02:
hello everyone, the instructions from @gpambrozio worked for me but I had to move `motion.sh` in `/usr/bin` and adjust the `motioneye.conf` accordingly. Hope this helps somebody :-)

Nuuki9, 2024-01-10:
I previously tried Bookwork using instructions from @gpambrozio and had some success, but autofocus and changing settings didn't work, and I don't think motion detection did either.

Based on comments from @JustGlowing I've just tried Buster but I don't see any cameras listed at all when I select `Local V4L2 Camera`. 

Clearly there are a lot of approaches being taken and all the work being put in is really appreciated. However with such a long thread and with differing methods its hard to nail down which ones work and with what caveats. Would it be feasible for us as a group to briefly write up any working methods in the wiki (or elsewhere), so we have a set of instructions that can be maintained and updated?

I bought the parts to make my mother a birdbox for her 80th, and so far its still on my workbench not working. As spring approaches I'd love to have this up and running for her.

JustGlowing, 2024-01-10:
@Nuuki9 under buster the camera is listed under MMAL Camera, not V4L2 Camera.

Nuuki9, 2024-01-10:
> @Nuuki9 under buster the camera is listed under MMAL Camera, not V4L2 Camera.

I'm pretty sure I checked that.

Currently Im testing the method @Peuz85 provided - that's actually working perfectly, expect the focus (as noted). I'll try adding the manual focus buttons and testing the motion capture, but hopefully it'll end up working.

MichaIng, 2024-01-23:
> under buster the camera is listed under MMAL Camera, not V4L2 Camera

If you enable KMS (`dtoverlay=vc4-kms-v3d`), it is listed under V4L2. This is just the default since Bullseye. However, that way it is not working without `libcamerify`.

Lacking time to look into this right now: Does someone know whether this really is a `motion` issue, simply not supporting the new libcamera API, so that `motionplus` is required, or is there something else we can do our end to make it functional with `motion` without an API wrapper/emulator like `libcamerify`?

[16 more replies]

closed 2026-11-13 by MichaIng