motioneye-project/motioneye #2684: working on a libcamera workarround

GenerallyClumsy, 2023-01-12:
I'm trying to get some cameras working that only support libcamera.  I thought I'd create a stream with libcamera-vid and add a network camera under motion eye. I've managed to create a network stream that's accessible under vlc. The command I'm using the create the stream is:
libcamera-vid -t 0  --width 1920 --height 1080 --inline --listen -o tcp://0.0.0.0:8888

while this stream works fine in vlc it doesn't work in motioneye 

FYI this is the URL used to connect to the camera in VLC tcp/h264://192.168.0.228:8888

On another note, the libcamera stream is working incredibly well. No dropped frames and CPU utilisation is low too. PI 3A

zagrim, 2023-08-23:
Libcamera support is still "under construction", and that applies to the yet unreleased ME 0.43 with Python3 support. The topic has been discussed here: #2425 (and here: #2683).

As far as I've understood things correctly, there are some workarounds mentioned in #2683, but no complete solution yet. There is a PR (https://github.com/motioneye-project/motioneye/pull/2765) linked to the other issue and it is waiting for testing, with some instructions on how that could be done (but I admit they might not be good instructions in case one is not that tech-savvy).

JamBalaya56562, 2026-11-06:
Running motion through `libcamerify`, as you found, is now the documented way: see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack). Checked with 0.44.0 on a Raspberry Pi 5 with a Camera Module 3 and motion 4.7.1, not on a Pi 3A; the page notes it works only on 64-bit systems. As for the stream, motionEye answers `protocol tcp not supported` for a `tcp://` URL, although motion itself can open it. Native libcamera support is tracked in #2812, so closing this as a duplicate of it. If the camera still does not work with these steps, please reopen.

closed 2026-11-06 by JamBalaya56562