motioneye-project/motioneye #2685: add camera with double /

manustars, 2023-01-12:
Hi to all, i have a problem, when i add new camera with this link rtsp://ipcamera:554/1/1/ (work on vlc) recived the error "An error occurred. Refreshing is recommended."

if change the link (this is wrong link) rtsp://ipcamera:554/1/ not recive any error but this not correct link

manustars, 2023-03-31:
any update for this?

JamBalaya56562, 2026-11-12:
There was no reply to the question about the version, so I am closing this. In the current release, the path of an RTSP URL, such as `/1/1/`, is kept as it is when the camera is tested and when the URL is passed to Motion. If it still fails for you on the current release (see https://github.com/motioneye-project/motioneye#installation), please open a new issue with the motionEye and Motion versions and the log.

closed 2026-11-12 by JamBalaya56562