motioneye-project/motioneye #2686: Embedded stream-iframe with no surveillance password set should not ask for a password

seb2010, 2023-01-14:
I do have setup multiple cameras and need to show the stream-iframe in a html-dashboard in the event of (external mesaured) motion.
Before I updated to the current version
```
motionEye Version | 0.43.0
Motion Version | 4.4.0
OS Version | Raspbian 11
```
(I am not sure what the old version was) I was able to access the iframe in the dashboard without beeing asked for a username/password via div-popup.

Now, after the update, I am asked for username/password every time I open the iframe, although no surveillance password is set/needed. Just clicking "Login" does the Job.
This is very much annoying and prohibits the use-case of just using the stream from motion/-eye to show on a html-dashboard. There cannot be any user interaction.
The "current-snapshot" does not work, as the frame script is the only way to show a video-like stream. Image refreshing comes with slower rendering times somehow.

As I specifically do want unauthorized surveillance access, there is no need to display the popup. If this is due to the  fact, that you like to be able to login with admin credentials to then configure within the iframe, a login icon for this would be better in that regard.

So the solution from my point of view would be:
- check if surveillance password is set. if not, show no div-popup for login in the iframe!

KR, SEB

seb2010, 2023-01-28:
For anyone eager to solve this issue but does not want to wait for a reaction here:
you need to alter the main.html template of motioneye, which is used to create the html-frame-page. On my dist this was found here: /usr/local/lib/python3.9/dist-packages/motioneye/templates/main.html

At the bottom of the template, you will find the definition of two DIV-containers "modal-glass" and "modal-container", which will show the login-screen on page load. You can selectively disable this for the frame by adding this line above the DIV-definition:
`{% if not frame %}` and `{% endif %}` below these two. Save the file and the login-popup will not show.

MichaIng, 2023-01-28:
Many thanks for reporting. Also no password in a way is a password, at least the username needs to be entered correctly, and you need to explicitly leave the password field empty.

We need to check how this was done on v0.42.x, whether it was intended, and whether the changed behaviour was intended. But year, as long as there is no option to explicitly allow a camera to be viewed without login, it is probably best to use an empty surveillance user password as indicator that this is wanted.

Btw, isn't it possible/better to frame the motion stream directly, instead of the motionEye UI?

seb2010, 2023-01-28:
Yes, somewhat true what you are saying. But let's be honest, just by overlaying a div above the still viewable content does not provide any security at all.

And please keep in mind, that motioneye is one of the options for rtsp camera owners to convert to mjpeg streams. So the ONLY use case for some might be just to access the plain stream. And since this is used to be displayed in smart home frontends, the options for NO authentication whatsoever needs to be available.

I actually use the frame, because it somehow is able to display the jog sequence more smoothly than a page with frequent refreshes.

MichaIng, 2023-01-28:
> But let's be honest, just by overlaying a div above the still viewable content does not provide any security at all.

That is true indeed 😄. I wonder whether the stream is also shown (below the div) if an actual password is applied, or whether this is indeed depending on empty vs non-empty password. Makes of course sense to align both.

seb2010, 2023-01-28:
Yes, the stream is playing. I just found the solution by paying around with Chrome Dev-Tools and just deleted the div. Below this, everything works as desired and prior to entering the pwd.

schnudd31do3, 2024-09-02:
Hi [seb2010](https://github.com/seb2010), you saved my life :-;
In the current version they have added one more DIV container. So currently you have to wrap all three DIV containers: "modal-glass", "modal-container" and "popup-message-container".

If they would not realize this feature, I alternatively suggest creating the option for own templates so that every update of motioneye does not overwrite the changes in the modified main.html.

Demetriao, 2025-03-22:
So this issue seems to have arisen after this PR: f6c9a2f.
Everything was still working fine in the Python 2 releases.

https://github.com/motioneye-project/motioneye/blob/f6c9a2f0cf1e482955eb53faccb91cbcf8490f8f/motioneye/static/js/main.js#L5484-L5491

It seems that the runLoginDialog function is always called when no auth cookies are set.

Maybe we should check whether a normal user password is present, and if not, just call doAuth() right away?

The backend appears to already accept requests when no normal user password is provided:

https://github.com/motioneye-project/motioneye/blob/8bf9bb9809a4fab0564f6a2526bf668f65f22c3c/motioneye/handlers/base.py#L149-L151

Currently it looks like this:
[image]
When you click Cancel or Login in the login popup without entering credentials, the popup just disappears.
So I don't think it's necessary to show this popup at all in such cases.

Demetriao, 2025-03-23:
I did some testing and this can easily be fixed with this change:
https://github.com/motioneye-project/motioneye/blob/f6c9a2f0cf1e482955eb53faccb91cbcf8490f8f/motioneye/static/js/main.js#L5484-L5491

to:
```
 if (isAuthCookiesSet() || frame) {
     doAuth(); 
 } else { 
     runLoginDialog(function () { 
         window._loginRetry = true; 
         doAuth(); 
     }); 
 } 
```
If someone has some time, could you please give some feedback on this?
Thanks

@MichaIng?

Demetriao, 2025-04-06:
I did some testing and this can easily be fixed with this change:
https://github.com/motioneye-project/motioneye/blob/f6c9a2f0cf1e482955eb53faccb91cbcf8490f8f/motioneye/static/js/main.js#L5484

to:
```
 if (isAuthCookiesSet() || frame) {
```
If you want to try this use this command:

`sudo sed -E -i 's/^[[:space:]]*if\(isAuthCookiesSet\(\)\) \{/    if(isAuthCookiesSet() || frame) {/' /usr/local/lib/python3*/dist-packages/motioneye/static/js/main.js`

You may need to adjust the path..

holdit, 2025-04-07:
Was having this issue after reinstalling MotionEye (was running an older version before) and @Demetriao change above seems to work well.

Demetriao, 2025-04-24:
Can this fix be added to the next beta release? @MichaIng 
> if(isAuthCookiesSet()) { 
> to:
> if (isAuthCookiesSet() || frame) {

holdit, 2025-05-05:
Just to add to my previous comment, @Demetriao's fix has been working fine with the direct link to the feed (`:8765/picture/1/frame/` in my case).

However, the page where we can review saved videos, pictures, etc, still shows the login form even when we don't have a login. We just have to click on "login" and it goes away, but still, I believe before this wouldn't happen.

[image]

The right thing to do would be to have a login, but this is only available on a local network and it's for my parents, so I have to keep it simple.

Demetriao, 2025-05-06:
Do you mean the main page at {ip_address}:8765?

With the current code, this isn’t possible, because once the login dialog is hidden, it's no longer possible to switch users.
As mentioned in a previous comment, a separate button would be needed for that.
However, since user interaction is more likely on this page, enabling the "Remember me" option in the login dialog might be a simple solution.

The fix I provided above restores the behavior to how it worked in MotionEye version 0.42.

holdit, 2025-05-09:
@Demetriao Yes, that page. I think at some point we wouldn't see that login dialog if there was no password... but I'm working from memory and might be wrong.

It's not a big deal, but I mentioned it in case it's not supposed to appear when we don't have a password.