motioneye-project/motioneye #2687: Please add "vaapi" support

carsten-f, 2023-01-16:
Hello motioneye-Devs & Maintainer,

i have tried to add vaapi in motioneye files, for recording streams, like qsv but unfortunately this was not successful.

Since "motion" supports vaapi it would be very awsome if you could add vaapi to motioneye because it would be enable all users with AMD APUs to use hwaccelation for recording streams in motioneye like the Intel-Guys with QSV. 

**vainfo:**

`libva info: VA-API version 1.10.0`
`libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/r600_drv_video.so`
`libva info: Found init function __vaDriverInit_1_10`
`libva info: va_openDriver() returns 0`
`vainfo: VA-API version: 1.10 (libva 2.10.0)`
`vainfo: Driver version: Mesa Gallium driver 20.3.5 for AMD KAVERI (DRM 2.50.0, 5.10.0-20-amd64, LLVM 11.0.1)`
`vainfo: Supported profile and entrypoints`
      `VAProfileMPEG2Simple            : VAEntrypointVLD`
      `VAProfileMPEG2Main              : VAEntrypointVLD`
      `VAProfileVC1Simple              : VAEntrypointVLD`
      `VAProfileVC1Main                : VAEntrypointVLD`
      `VAProfileVC1Advanced            : VAEntrypointVLD`
      `VAProfileH264ConstrainedBaseline: VAEntrypointVLD`
      `VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice`
      `VAProfileH264Main               : VAEntrypointVLD`
      `VAProfileH264Main               : VAEntrypointEncSlice`
      `VAProfileH264High               : VAEntrypointVLD`
      `VAProfileH264High               : VAEntrypointEncSlice`
      `VAProfileNone                   : VAEntrypointVideoProc`

**motion log**
`[1:ml1:Camera-Cars] [DBG] [ALL] util_parms_parse: Parsen: keepalive=off,tolerant_check=on,decoder=vaapi`
`[1:ml1:Camera-Cars] [DBG] [ALL] util_parms_add: Geparst: >keepalive< >off<`
`[1:ml1:Camera-Cars] [DBG] [ALL] util_parms_add: Geparst: >tolerant_check< >on<`
`[1:ml1:Camera-Cars] [DBG] [ALL] util_parms_add: Geparst: >decoder< >vaapi<`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_set_parms: Norm Stream wird eingerichtet.`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_set_path: Einrichtung von rtsp über Netcam`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_set_options: Norm: Einstellung von rtsp/rtmp`
`[1:ml1:Camera-Cars] [DBG] [ALL] util_parms_add: Geparst: >rtsp_transport< >tcp<`
`[1:ml1:Camera-Cars] [DBG] [ALL] util_parms_add: Geparst: >allowed_media_types< >video<`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_set_options: Norm: Option: keepalive = off`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_set_options: Norm: Option: tolerant_check = on`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_set_options: Norm: Option: rtsp_transport = tcp`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_set_options: Norm: Option: allowed_media_types = video`
`[1:ml1:Camera-Cars] [INF] [NET] netcam_rtsp_open_context: Norm: Geöffnete Kamera ( Camera-XXXXXXXX)`
`[1:av0:Camera-Cars] [INF] [NET] netcam_hwdecoders: Norm: HW-Geräte: `
`[1:av0:Camera-Cars] [INF] [NET] netcam_hwdecoders: Norm: vdpau (nicht implementiert)`
`[1:av0:Camera-Cars] [INF] [NET] netcam_hwdecoders: Norm: vaapi (verfügbar)`
`[1:av0:Camera-Cars] [INF] [NET] netcam_hwdecoders: Norm: qsv (nicht implementiert)`
`[1:av0:Camera-Cars] [INF] [NET] netcam_hwdecoders: Norm: drm (nicht implementiert)`
`[1:av0:Camera-Cars] [INF] [NET] netcam_hwdecoders: Norm: opencl (nicht implementiert)`
`[1:av0:Camera-Cars] [INF] [NET] netcam_init_vaapi: Norm: Vaapi-Decoder initialisieren`

TheRaven500, 2024-05-31:
Why does motion support vaapi and motioneye not? Motioneye is "only" a GUI for motion or i am wrong? :thinking:

MichaIng, 2024-05-31:
Neither motionEye, nor motion have an impact on whether VA-API is supported or not. It is the host system and the underlying libraries only.

> Why does motion support vaapi and motioneye not? Motioneye is "only" a GUI for motion or i am wrong? 🤔

Exactly. motion indeed has separate functions for separate decoders when decoding network camera streams, but those contain nearly identical code, essentially calling the FFmpeg libraries: https://github.com/Motion-Project/motion/blob/master/src/netcam_rtsp.c#L325
FFmpeg is what really does the decoding, based on how it was compiled, which libraries the host system ships with, and what its hardware supports.

> i have tried to add vaapi in motioneye files

Which motionEye files do you mean? There are no options for this, neither in motionEye, nor in motion. This is all detected automatically by motion (resulting in the logs you see), and this detection does not even have any relevant impact, since FFmpeg finally does the decoding, using the matching decoder by itself, regardless what motion detected, AFAICS. I.e. those logs are informational only, not a decision/setting/option motion actually does/have.

And actually VA-API is used in your case, isn't it?
```
[1:av0:Camera-Cars] [INF] [NET] netcam_hwdecoders: Norm: vaapi (verfügbar)
[1:av0:Camera-Cars] [INF] [NET] netcam_init_vaapi: Norm: Vaapi-Decoder initialisieren
```

Since the issue is old, I'll mark this as closed. If someone still has issues with VA-API support on modern Debian (Bookworm and above), you may open a new issue. But it is probably more beneficial to ask this at some Debian and/or FFmpeg forum then.

closed 2024-05-31 by MichaIng

TheRaven500, 2024-06-01:
Yes i think you are right. In the case of OP it runs on vaapi. But in my case not:
```Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: HW-Geräte:
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: vdpau (nicht implementiert)
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: cuda (verfügbar)
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: vaapi (verfügbar)
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: qsv (nicht implementiert)
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: drm (verfügbar)
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: opencl (nicht implementiert)
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_hwdecoders: Norm: vulkan (nicht implementiert)
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_init_swdecoder: Norm: Initialisieren des Decoders
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_rtsp_open_codec: Norm: Decoder geöffnet
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: HW-Geräte:
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: vdpau (nicht implementiert)
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: cuda (verfügbar)
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: vaapi (verfügbar)
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: qsv (nicht implementiert)
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: drm (verfügbar)
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: opencl (nicht implementiert)
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_hwdecoders: Norm: vulkan (nicht implementiert)
```
As you can see "vaapi" is available.
But:
```
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_init_swdecoder: Norm: Initialisieren des Decoders
Jun 01 10:29:41 MotionEye motion[632]: [2:av0:Camera2] [INF] [NET] netcam_rtsp_open_codec: Norm: Decoder geöffnet"
``` 
No clue why, but it is a OS problem and not a motion or motioneye problem. Looks like ffmpeg on debian is missing vaapi i think.

MichaIng, 2024-06-01:
> As you can see "vaapi" is available.
> But:

In the two log lines I do not see which decoder is finally used. But makes sense that it is not VA-API when this FFmpeg build does not support it.

> Edit2:
> Camera.conf was wrong, now it tries to enable vaapi but it does not working:

What do you mean by "was wrong"? Unless you added mentioned decoder override manually, it does/should not enforce one. Or do you mean you tried to enforce it wrong, at first, and now you used the correct syntax? Then we know that it indeed can be enforced, and that your FFmpeg does not support it.

> Looks like ffmpeg on debian is missing vaapi i think.

It also depends on the platform/hardware. VA-API was an Intel GPU feature at first, and while newer other architecture GPUs started to support it, or have wrappers for it, on e.g. ARM SBCs there is no assured mainline support across all applications.

TheRaven500, 2024-06-01:
If there is no parameter, motion is using "swdecoder". You can see it in the log:
```
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_init_swdecoder: Norm: Initialisieren des Decoders
Jun 01 10:29:41 MotionEye motion[632]: [1:av0:Camera1] [INF] [NET] netcam_rtsp_open_codec: Norm: Decoder geöffnet
```
If i try to force vaapi with parameter "decoder = vaapi", then it tries to enable vaapi but it does not work.
Maybe it is a hardware problem. It is an "old" core2duo with a Radeon HD2400 (RV610) graphic card. Not sure if the radeon is enough for vaapi or if it needs a newer cpu. My plan was to reduce cpu usage by adding a radeon graphic card. But it looks like this plan does not work :wink:

MichaIng, 2024-06-01:
I guess software decoding is only used as default in your particular case for this particular netcam stream/format. FFmpeg surely uses hardware decoder, where available.

motion reads the used decoder from some memory byte. I didn't have a closer look, but I guess it asks FFmpeg what it will use, then calls the related function you see in logs. Those functions however contain all moreless the same code and call the FFmpeg API without additionally defining which decoder should be used, AFAICS.

TheRaven500, 2024-08-08:
Sorry for late response. I was far away :wink: 
I tried a lot of things (decoder = h264_qsv for example) but i don't get vaapi to work. I tested with several radeon cards and with an i7 (Intel Skylake Gen9) but all of them use swdecoder. I see high cpu load and if i run "radeontop" or "intel_gpu_top" there is 0% of load. I think this is a motion and not a motioneye problem but i am not sure. No clue how to enable HW decoding. :disappointed: