motioneye-project/motioneye #2690: nvenc missing in docker

klyoku, 2023-01-22:
Running motioneye docker. Everything works well, but hw encoding option seems to be missing:
[image]

nvidia driver is loaded in the container:
[image]

Not sure if I missed a step/config, or if this is a bug. Any pointer to what log I need to pull?

shadmanhabib, 2023-02-14:
Hey @klyoku. Sorry I don't know the answer to your question but I think you can answer a question that I have.

Which docker image did you use to run motioneye? Yours seems to have QSV support but mine does not.

klyoku, 2023-02-14:
this:
https://github.com/motioneye-project/motioneye/blob/dev/docker/docker-compose.yml