motioneye-project/motioneye #2692: SMTP AUTH extension not supported by server.!

DJAlexOfficiel, 2023-01-25:
Hello,

In the motion notifications tab, I entered the information of my outlook account but I got this error. My id and mdp are corect.

Notification email failed: SMTP AUTH extension not supported by server!

Server name: smtp.office365.com
Port : 587
Encryption method: STARTTLS

martinL44, 2023-05-05:
i have the same problem

JamBalaya56562, 2026-11-12:
Closing, as there was no follow-up to MichaIng's suggestion. motionEye switches to TLS with STARTTLS and then logs in with the account and password (`motioneye/sendmail.py:76-79`). This error means the server did not offer password login after that, which depends on the mail account and the provider: for Outlook, SMTP AUTH has to be allowed for the mailbox. Please open a new issue if this still happens on a current release with a server that allows SMTP AUTH.

closed 2026-11-12 by JamBalaya56562