motioneye-project/motioneye #2699: live video stopped working in current browsers

vcaputo, 2023-02-03:
I have some RPi Zero W cams running motioneye I setup a couple years ago.

Everything worked great for ages but it seems at some point by keeping my desktops/laptops current, the latest browsers don't work with whatever MotionEye is using for the live video feeds.

These cameras are very slow and they're installed at a remote off-grid location having limited bandwidth.

Is this an understood problem with an easy spot-fix I can do on the cameras to get the live feed back?

Everything is still working, and I am still able to access the still images in the archives as well as receiving notifications.  But when I'm on-site it's quite useful to bring up the live feeds when I get a motion notification, and that no longer works.

Thanks in advance for any assistance, and apologies if this is a misuse of the issues section.  If there's a preferred space for such questions please refer me there and I'll close this!

Motioneye has been really great, it's just too bad the web seems utterly incapable of constantly breaking itself.

zagrim, 2023-07-03:
I don't know if you already ended up doing that, but "several years" old ME setup sounds like it has everything dated enough that it just might be that the current browsers have changed so that whatever the old video stream did is not anymore compatible. For example, we had this: #1428. Does that match what you are experiencing?
Unless that works, and unless you are able to find here a similar issue with a workaround, upgrade is likely the easiest solution.

vcaputo, 2023-07-03:
Thanks, that indeed is the browser change affecting my cams.

closed 2023-07-03 by vcaputo