motioneye-project/motioneye #2701: FreeBSD 13.1 Meyectl fails in python 3.7 and 3.9, no module named "settings".

aredon, 2023-02-10:
I upgraded from freebsd 11.3 to 13.1 today and was updating all my packages. I decided to revert to python 3.7 instead of 3.9 as I knew that worked previously but meyectl won't run and thus motioneye will not start. I have already ran pip install --upgrade motioneye.

[image]

zagrim, 2023-02-10:
You need yet-officially-unreleased Python 3 compatible version of ME to run with Python 3. Please see the installation instructions at https://github.com/motioneye-project/motioneye/blob/dev/README.md . You might need to adapt some things for FreeBSD, e.g. if you don't have `python3` there just substitute whatever runs stuff on Python 3, e.g. if `pip` is of Python 3, you might do simply `sudo pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'`

aredon, 2023-02-10:
I was previously running python3.7 on freebsd 11.3 with motioneye running fine. Do I need that specific tar over `pip install motioneye` or `pip install --upgrade motioneye` ?

zagrim, 2023-02-10:
I don't know about FreeBSD 11.3, as I'm a Linux user, but earlier releases of several Linux distros had Python 2 as the default version (as `python` and `pip`) while still also having Python 3 (as `python3` and `pip3`), and that error you have there is just what you get when trying to run the old Python 2 codebase on Python 3.

As for the actual command, I'm not sure if the upgrade would work ok since the changes are pretty substantial and I'm not an expert on how `pip` manages things. You might give it a try of course, but if that somehow fails, a clean install should do the trick.

aredon, 2023-02-11:
Alright I installed that tar ball and here's where we are now. The motioneye service is running (I can reach the web page) but the cameras are blank and motion seems to not be detected. I tried running `motioneye_init` and got this message and then I checked whether or not I had the missing dependencies and I do.

[image]

zagrim, 2023-02-12:
What on earth is that `motion` 0.2.0 that is installed via `pip`? Doesn't look like the right dependency at all. What you need instead is this one: https://motion-project.github.io/

aredon, 2023-02-12:
Ha! That's something isn't it. I removed motion via `pip uninstall motion` then I tried their build instructions for FreeBSD and this happened. I also tried `make clean` and then `make` again with the same result.

https://motion-project.github.io/motion_build.html
[image]

zagrim, 2023-02-13:
I'm pretty far off my expertise area here, but it looks like you might be missing a shared library, or perhaps the build scripts (Makefile) of Motion isn't quite compatible with FreeBSD. Do you have `/usr/local/lib/libsharpyuv.so`? Or that file in some other path?
I was also wondering if [this](https://docs.freebsd.org/en/books/handbook/linuxemu/#linuxemu-libs-manually) might be relevant, but then again if you've installed all the dependencies from FreeBSD package repos, I doubt it is not relevant here. Nevertheless, I do not have that .so file on my system, but I've also not compiled motion from source. No Debian packages also contain file named like that, so I really don't know what's this about.

You might be better off searching for help on compiling and/or installing `motion` on FreeBSD from the motion-project or some other FreeBSD users. Since others have successfully installed ME on FreeBSD, I'm pretty confident that after you get this issue sorted out things would go nicely from there on.

aredon, 2023-02-13:
Unfortunately I do indeed have the /usr/local/lib/libsharpyuv.so file present on my machine. It would seem that it is missing functions that motion is expecting it to have. I haven't had a lot of luck figuring out what package even carries that library so I can check the version either. 

I already compiled from motion-project per their instructions. No such luck unfortunately. 
[image]

I will open a thread over on the motion-project git page and hopefully get some answers. When I figure it out I'll post here just for posterity.

aredon, 2023-02-18:
Alright. It seems that webp is not correctly configured on a fresh freebsd install. So motion must be compiled without it.

https://github.com/Motion-Project/motion/discussions/1641#discussioncomment-5015769

tehfink, 2023-02-20:
You might try the working setup outlined in this iocage plugin for motioneye on FreeBSD 13, see `post_install.sh`: https://github.com/Baenwort/iocage-plugin-motioneye

closed 2026-11-01 by MichaIng