motioneye-project/motioneye #2702: Camera overlay is not visible when the page loads, but the invisible buttons are still clickable

af7567, 2023-02-14:
When loading the main motioneye page the buttons along the top are not visible and it requires 2 clicks to get the overlay to show. It seems like motioneye thinks the buttons should be visible, but the UI doesn't. The invisible buttons can still be clicked.

By adding `visible` to the class when creating the cameraFrame it works as expected with the initial state visible and clickable.

https://github.com/motioneye-project/motioneye/blob/268c8a6ba2d9a5ebb29b421488321972f9777adf/motioneye/static/js/main.js#L4647

Since I prefer to have the buttons visible on load then this fix works for me, but maybe the intention was that they are initially invisible and one click should show them?

zagrim, 2023-02-16:
I'm not quite sure what you mean, but I assume you mean these buttons in the top right of camera frame:
[image]

I don't remeber ever having to click more than once on the frame to get the buttons visible, and if I click on a spot where the button would be while it is hidden, I just get the buttons visible and the click is not registered as if I had clicked the button. I just tested this on both ME 0.41rc1 (which I really should update some day...) and the current dev branch (v0.43), using rather current Chromium on desktop.

> maybe the intention was that they are initially invisible and one click should show them?

Exactly. The buttons and the top and bottom banners cover part of the frame, which is not nice if you just want to see what there is in the picture.

What kind of setup you have? Which Motioneye version? What browser?

af7567, 2023-02-17:
Yes, those are the buttons. I am using the current dev branch, I haven't used the python 2 version for a long time so can't remember if it happened on there.
I have tested in Firfox and Chromium and get the same thing. I just reinstalled todays dev branch so it is up to date and removed any changes I would have made when testing stuff.

I tried to record a little video showing what I meant. In the video I reload the motioneye page with CTRL-F5 and it shows the video without the buttons but when moving my mouse in the button area the tooltips can be seen and clicking the invisible picture browser button opens it.
I then CTRL-F5 again and have to click twice to make the buttons be visible (of course you can't tell how many times I am clicking from a silent video :) )

https://user-images.githubusercontent.com/12093847/219646733-e62d436e-910e-4b35-ad2e-7e5d4b0ae92f.mp4

zagrim, 2023-03-05:
Sorry for the delay, but I did somehow manage to reproduce this with current dev branch. I'll try to pinpoint what actually triggers this, since it does not happen with all layout related settings for me.

af7567, 2023-03-07:
It looks like something to do with the timing of the call to `hideCameraOverlay` when the page is created. I can see that the overlay is created visible but with opacity 0, so even though I can't see the buttons the click callback is intercepted by the overlay which calls `hideCameraOverlay` again.  

Changing this line:
https://github.com/motioneye-project/motioneye/blob/ae03a4c394d8ab2b8792636b3388bb4375ac2792/motioneye/static/js/main.js#L5113-L5114
to
```
setTimeout(function () { hideCameraOverlay(); }, 500);
```

so that the page is given a chance to be created before hideCameraOverlay is called seems to work.

zagrim, 2023-03-11:
Created PR with a fix. The simplest to me was to just enforce the overlay be hidden (`display: none`) from creation.

zagrim, 2023-03-11:
And thanks for the pointer @af7567 , that helped a lot in figuring this out.

af7567, 2023-03-11:
> Created PR with a fix. The simplest to me was to just enforce the overlay be hidden (`display: none`) from creation.

Yes, that makes much more sense than hiding it after creation. I have tested it here and it works fine on my PC and tablet.

MichaIng, 2023-03-12:
Solved in `dev` 👍.

PR #2717 (2023-03-12): Fix camera-overlay div sometimes being clickable (but invisible) on creation
Fixes #2702 .

There seemed occasionally to be race-condition between the creation of camera overlay in DOM and setting it hidden by `hideCameraOverlay` in `recreateCameraFrames` here:
https://github.com/motioneye-project/motioneye/blob/268c8a6ba2d9a5ebb29b421488321972f9777adf/motioneye/static/js/main.js#L5011

I left that call to `hideCameraOverlay` on line 5114 there for defensive purposes just in case, even though I think it currently is not needed. But if someone adds some more things to happen on show/hide of cam [... cut]
closed 2023-03-12 by MichaIng