motioneye-project/motioneye #2708: Telegram motion notifications not working

DeadlySin2, 2023-03-02:
I have the following settings of my Motioneye
[image]
[image]
I'm trying to make motion notifications working but they refuse to. My Home Assistant is detecting motion from a curl command, so as i think motioneye is able to detect motion. 
If i click Test button below Telegram settings i recieve test message to my Telegram bot, but the are no notifications about motion detected and no still images. 
My motioneye is installed in docker container from this repo:

motionEye Version | 0.43.0
Motion Version | 4.5.1
OS Version | Linux 4.4.180+

Please help to make notifications work. Thanks for your support.

s17534, 2023-03-04:
... but you have set:
Capture mode in Still images section to manual, so until you manually press snapshot button AND if time which passed from start of the motion detection to that moment of snapshot is not greater than Attached Images Time in Telegram settings section it should work.

So imo you should set Capture Mode to Motion Triggered [ One Picture ] and adjust Attached Images Time accordingly.

Linked above Telegram optimizations is another way of dealing with that issue.

zagrim, 2023-07-02:
I think @s17534 is correct about the capture mode of still images. @DeadlySin2 did you get any notification even without pictures? I think you should get, even if there are no images found, but how about if you change that mode?

DeadlySin2, 2023-07-03:
No I can't manage it working, so i just made some motion sensors in my Home assistant and recieve notification from there.

zagrim, 2024-03-20:
I do agree that the help/tooltip text for "Attached Images Time" is not as good as it should be, but I do not think there's much mystery in how that works (or at least how it should work based on code): The specified time span should be long enough that images captured during the motion event can be found within that time span (counting backwards from end of the event).

So, the suggested default 30 seconds means that any images saved within the previous 30 seconds should be attached. Whether there are any saved images depends on how long the motion event is, and if still images are saved for every motion frame or only for the "best" frame selected by Motion. The length of the motion event cannot be known except for "Maximum Movie Length" under "Movies" if that is specified.

Frame count related settings effect motion detection, but I don't immediately see how they could interfere with image attachment to notification... "Maximum Movie Length" can, in case of long max length and still image capture mode of "Motion Triggered (One Picture)" together with short "Attached Images Time".

Perhaps some useful validation between settings could be done, but in general that might quickly go pretty complex and not very stable :thinking:

aelg305, 2025-05-13:
anyone able to figure this out?

I can confirm pictures been taken and saved to the folder
No delivery of photos into telegram.... :(
I tested using settings for "still image": manual, motion, motion 1 photo, havent tried interval or all frames....
I also tested setting in telegram setting "attached image time": 0, 1, 5, 30..

Any ideas? 

[image]

aelg305, 2025-05-28:
You did get a notification?!? Niceee...I don't even receive those!

What do the logs say?

ameisenbr, 2025-06-20:
> Is there anyone here who received a notification to Telegram? I tried all combinations of Still Images and Motion Notification but didn't get any image to my Telegram .... I get only test message from my motioneye. It looks like "Notification to Telegram doesn't work". Can anyone show the settings where Motion notification sends a photo to Telegram ?

It works, though how many images get sent seems a bit arbitrary. One time it didn't send any. At least the first image seems to be right from the moment of first detection, which is good enough for me. I had to turn off simultaneous moive capture, it made it work unreliably and the movies had a long freeze period once the API call got made.
My setup:
Pi 4B 2GB, RaspberryPi Cam v3
motionEye Version 	0.43.1b4
Motion Version 	4.7.0
OS Version 	Debian 12
(bookworm 64 bit lite)
[This](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345) excellent post helped me to get it working with libcamera

[image]

[image]

[image]

SfashTheFirst, 2025-08-31:
Похоже, я понял, как заставить уведомления работать. Они присылаются только если будут выполнены 2 условия:
1. В папке с фотографиями есть фотография, которую можно прикрепить к этому уведомлению (т.е. сделана в интервале, который вы указали как "attached images time" в "motion notification)
2. В этой папке не очень много фото. У меня всё перестаёт работать если их больше 1000

Как решить первое - понятно (смотри раздел "still images"). 

Второе я решил очисткой папки с фото каждый час на уровне ОС по таймеру:

**Файл скрипта**: "nano /usr/local/bin/cleanup_photos.sh"

`#!/bin/bash

BASE_DIR="your path /data/motioneye/Camera1" #YOUR PATH TO FOLDER WITH PHOTOS FROM MOTION
CURRENT_DATE=$(date +%Y-%m-%d)
CURRENT_HOUR=$(date +%H)
TARGET_DIR="${BASE_DIR}/${CURRENT_DATE}"

if [ ! -d "$TARGET_DIR" ]; then
    exit 0
fi

cd "$TARGET_DIR" || exit 1

LATEST_FILE_TO_KEEP=$(
    find . -maxdepth 1 -type f -name "*.jpg" -print0 | \
    while IFS= read -r -d '' file; do
        filename=$(basename "$file")
        file_hour=${filename%%-*}  
        if [ "$file_hour" -lt "$CURRENT_HOUR" ]; then
            echo "$file"
        fi
    done | sort -r | head -n 1
)

if [ -z "$LATEST_FILE_TO_KEEP" ]; then
    exit 0
fi

find . -maxdepth 1 -type f -name "*.jpg" -print0 | \
while IFS= read -r -d '' file; do
    filename=$(basename "$file")
    file_hour=${filename%%-*}
    if [ "$file_hour" -lt "$CURRENT_HOUR" ] && [ "$file" != "$LATEST_FILE_TO_KEEP" ]; then
        rm -- "$file"
    fi
done`

2. Делаем исполняемым "chmod +x /usr/local/bin/cleanup_photos.sh"
3. Ставим скрипт в systemd "nano /etc/systemd/system/photo-cleanup.service"
`[Unit]
Description=Очистка старых фотографий камеры наблюдения

[Service]
Type=oneshot
User=root
ExecStart=/usr/local/bin/cleanup_photos.sh`
4. Делаем таймер "nano /etc/systemd/system/photo-cleanup.timer"
`[Unit]
Description=Запуск очистки фотографий в первую минуту каждого часа
Requires=photo-cleanup.service

[Timer]
OnCalendar=*-*-* *:01:00
Persistent=true

[Install]
WantedBy=timers.target`
5. Включаем 
`systemctl daemon-reload
systemctl enable photo-cleanup.timer
systemctl start photo-cleanup.timer`

MichaIng, 2026-10-23:
Reviewing this issue:
* Of course still images need to be enabled with something other than "manual", else no pictures are created during the event, hence nothing can be attached.
* If set to "motion-triggered" without just one picture, a rather short "attached images time" should do. There will be a certain hardware-depending delay until motion starts to store pictures, but a too long times means potentially too many attached pictures.
* If set to "one picture", the "attached images time", again, should be possible rather short, like 5 seconds or less, since the first and only picture should be saved quite shortly after event start.
* If set to "best", then the "attached images time" need to be longer than the whole event + some buffer for motion to process all frames, find the one with most changes, generate and store the image. Also take into account "post-capture" and the default 30 seconds event gap, which add on top. So the "best" still image method does not play well with notifications, as you have no control about how long an event takes, and need to use a very long "attached images time" (hence notification delay) to raise the change to get any picture.
* If motion-triggered movie recording is enabled, too, then the delay until motion stores pictures might be longer, even a lot longer, depending on the used hardware, since, of course, CPU processing and disk I/O for movie and pictures overlaps.

In any case, what helps to debug things and find out the exact time motion takes to send an event start API call, the picture safe API call, and how long motionEye in case takes between triggering the Telegram notification, and the Telegram being sent out (which does include the "attached images time"), can be checked with debug logs enabled via `journalctl -u motioneye`. motionEye v0.44.0 also enhanced the time it takes for media files to be listed, and v0.45.0 will enhance it again a lot. That does not affect the number of collected pictures, but reduce a potentially additional significant delay until the notification is actually sent out.

free-cup-holder-exe, 2026-11-08:
For reference, the 0.45 changes MichaIng mentioned, all in 0.45.0b1:

- #3425: Telegram lists only the pictures inside the attached-images time. A folder with thousands of pictures no longer makes the listing time out, which fits what SfashTheFirst found at about 1000 photos.
- #3413: a notification whose picture listing fails is still sent, without pictures, and the log names the timeout setting.
- #3457: a request Telegram rejects (wrong chat id or token) is logged as failed, and the Test button says so. Before, both said sent.
- #3511: one picture that fails no longer stops the others.

The capture-mode points above still apply: with Manual still images there is nothing to attach.

JamBalaya56562, 2026-11-14:
Closing this, as 0.45.0 has the fixes linked here: Telegram now lists only the pictures within the attached images time (#3425), sends the notification even when that listing fails (#3413), checks Telegram's reply and retries once (#3457), and keeps sending the other pictures after one fails (#3511). Pictures are only attached when still images are captured during the event, as described above (not with *Manual*). If notifications still do not arrive with 0.45.0, please reopen with a debug log from `journalctl -u motioneye`.

closed 2026-11-14 by JamBalaya56562