motioneye-project/motioneye #2709: Can not start Motioneye after reboot.

Xadro3, 2023-03-03:
Hi, I've installed Motioneye on a pi0 yesterday and after a lot of trial and error managed to install it on raspbian buster, I followed every step in the wiki and managed to get it to run. I turned the Pi off and mounted it outside again and turned it on, but Motioneye did not start, manually starting it with 
systemctl daemon-reload
 systemctl enable motioneye
 systemctl start motioneye
Doesn't do anything either.

Xadro3, 2023-03-03:
I get no error messages, I run as root

MichaIng, 2023-03-03:
Please try these instructions to install the latest Python 3 compatible version: https://github.com/motioneye-project/motioneye/tree/dev#installation

Xadro3, 2023-03-03:
I followed the Python 3 instructions, and i can see motion now being active, but i can not see any process pertaining to the webinterface, i can not connect to it via browser either.
[image]

Xadro3, 2023-03-03:
[image]
seems like a permission issue or maybe the folder was not setup?
[image]
motion is running, but throwing permission errors for ffmpeg

MichaIng, 2023-03-03:
Okay, first of all disable motion:
```sh
systemctl disable --now motion
```
And then I think you have an old motionEye service in place:
```sh
systemctl cat motioneye
```
Please try to replace it with the new one:
```sh
curl -sSfo /etc/systemd/system/motioneye.service 'https://raw.githubusercontent.com/motioneye-project/motioneye/dev/motioneye/extra/motioneye.systemd'
systemctl daemon-reload
systemctl restart motioneye
sleep 3
journalctl -u motioneye
```
The installer is conservative and does not replace an existing unit, but this causes more issues than it solves.

Xadro3, 2023-03-03:
I followed your Instructions and i got some permission issues, i went into each directory to give the permissions manually, but i now get 
[image]
a smb error which i am not sure how to fix

MichaIng, 2023-03-03:
If you do not use it, disable the SMB mounts option: https://github.com/motioneye-project/motioneye/blob/ad21146/motioneye/extra/motioneye.conf.sample#L56-L57

I opened another issue about enabling this feature in non-root mode, which is generally possible: #2710

closed 2023-03-03 by Xadro3