motioneye-project/motioneye #2710: Enable SMB mounts in non-root mode

MichaIng, 2023-03-03:
This is generally possible with proper group membership and/or capabilities.

Xadro3, 2023-03-03:
I did grant the user motion root access, i think. How would i go about starting motioneye in root mode?

Xadro3, 2023-03-03:
I edited the passwd file to change the GID of motion user to root and i do see a start now, thank you for your help!

JamBalaya56562, 2026-11-14:
@MichaIng I looked into how this could work. `mount.cifs` itself refuses a non-root caller unless the share has an fstab entry with the `user` option and `mount.cifs` is installed setuid root (`check_setuid()` and `check_fstab()` in [mount.cifs.c](https://github.com/piastry/cifs-utils/blob/master/mount.cifs.c)), so capabilities for the service alone would not be enough as long as motionEye calls `mount.cifs`. Options I can see:

1. `motioneye_init` installs a small root-owned helper and a sudoers rule that allows only that helper for the motion user; motionEye calls it with `sudo -n`, and the helper only mounts and unmounts below `smb_mount_root`.
2. Call `mount(2)` directly from Python (ctypes, no new dependency), with `AmbientCapabilities=CAP_SYS_ADMIN` in the unit; this has to do what `mount.cifs` does (resolving the server etc.) itself, and `CAP_SYS_ADMIN` is close to root.
3. Keep requiring root, and document a drop-in with `User=root` for those who need SMB shares.

I would go with 1, as it changes the least and keeps what the motion user may do narrow. Which way would you prefer?

MichaIng, 2026-11-15:
Nah with `CAP_SYS_ADMIN` nothing is won. A pain that it is no difficult for non-root users to mount CIFS shares, since authentication happens internally, so the reasons for mount restrictions for local drives, as well as for NFS shares (often authenticated only per host, not per user), do not hold here.

Anyway, it is as it is. A `sudoers` rule could theoretically also cover the `mount`/`umount` command directly, without a helper. But it might be difficult to reliably match those with the mount point as 2nd positional argument. `mount -t cifs --target <smb_mount_root> //server/share -o ...` could be used, which allows a strict `sudoers` command prefix. But I am not sure how well portable this is, thinking about our recent FreeBSD support efforts. Also, if `smb_mount_root` is changed, the sudoers rule needs to change. So a wrapper is more robust, indeed.