motioneye-project/motioneye #2712: Question: There is a way to save red lines that show motion area in the video overlay?

Metus88, 2023-03-09:
When it is detected a motion and you are looking the web interface on a browser you can see rectangular or square red lines that delimitate the area that is on motion, there is a way to save also this lines on the "File Storage". Thanks a lot

zagrim, 2023-03-10:
I don't think that is possible to do. The red box is drawn by Motion ([`locate_motion_mode` and `locate_motion_style`](https://motion-project.github.io/motion_config.html#locate_motion_mode)), and while `locate_motion_mode on` is being set by ME when Show frame changes is enabled, and the docs kind of suggest that "on" would affect saved movies, it apparently doesn't.

Metus88, 2023-03-10:
Thanks a lot! Now I tested "locate_motion_mode on", (with and without text overlay) and like you said, no red box shown on the video file stored. Do you thinks is a bug?

zagrim, 2023-03-11:
Asking about it in [Motion project](https://github.com/Motion-Project/motion/issues) might at least lead to improved documentation in case it is not a bug. I did a quick search with "locate_motion_mode" in issues and didn't find anything that seemed to be really about this based on the issue titles.

Metus88, 2023-03-11:
Thanks for your help. I have just open a issue [there](https://github.com/Motion-Project/motion/issues/1648#issue-1619954210).

closed 2023-03-11 by Metus88