motioneye-project/motioneye #2714: Sunny days trigger motion

Xadro3, 2023-03-10:
Since the weather cleared up recently i had the issue that sudden increases of contrast between shadows and sunny patches trigger motion detection, a lot 
[image]
[image]
i do also believe this issue ignores working schedule and masking 
[image]
[image]
[image]

MichaIng, 2023-03-10:
The working schedule indeed has an issue currently we need to have a look into. The mask however worked well last time I tested it. Generally it makes sense and is a known limitation of motion, that sudden light condition changes, including clouds and such, trigger motion detection.

Xadro3, 2023-03-11:
Thanks for the info, i will try to workaround that limitation then, maybe try to get a motion detector going and use the client api to trigger motion/snapshots that way

JamBalaya56562, 2026-11-13:
The working schedule issue mentioned above was fixed in 0.43.1 (#2987, see #2815). Sudden light changes are a known limitation of Motion's detection, as @MichaIng said; *Motion Detection → Light Switch Detection* makes Motion ignore frames where a large part of the picture changes at once, and a mask can leave out the areas where shadows move. Closing; please open a new issue if the schedule or mask is still ignored on a current release.

closed 2026-11-13 by JamBalaya56562