motioneye-project/motioneye #2726: Trouble to get SMB Share added.

cryd-s, 2023-03-18:
Sorry guys but i have tried several things and was reading existing issues about this problem but cannot get a working smb share.

Someone an idea?

Here is my Config:

[image]

[image]

[image]

[image]

zagrim, 2023-03-19:
Please attach logs (`sudo journalctl -xu motioneye` if on system with systemd).

cryd-s, 2023-03-19:
`Mar 19 10:51:24 motioneye meyectl[10439]: /bin/sh: 1: lsb_release: not found
Mar 19 10:53:00 motioneye meyectl[10488]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10488]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10490]: mount error(22): Invalid argument
Mar 19 10:53:00 motioneye meyectl[10490]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10492]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10492]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10494]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10494]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10496]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10496]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10498]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10498]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[109]:    ERROR: failed to mount smb share "//192.168.178.153/nvr" at "/media/motioneye_192_168_178_153_nvr_administrator"
Mar 19 10:53:00 motioneye meyectl[109]:    ERROR: access to network share //192.168.178.153/nvr failed: cannot mount network share
Mar 19 10:53:27 motioneye meyectl[10506]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10506]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10508]: mount error(22): Invalid argument
Mar 19 10:53:27 motioneye meyectl[10508]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10510]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10510]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10512]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10512]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10514]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10514]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10516]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10516]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[109]:    ERROR: failed to mount smb share "//192.168.178.153/nvr" at "/media/motioneye_192_168_178_153_nvr_administrator"
Mar 19 10:53:27 motioneye meyectl[109]:    ERROR: access to network share //192.168.178.153/nvr failed: cannot mount network share
`

cryd-s, 2023-03-19:
with my windows i can connect to the folder with the same credentials

cryd-s, 2023-03-19:
https://pastebin.com/xPSBCqhf here is the log from several trys with different options

zagrim, 2023-03-19:
Well, there you have it:
```
mount error(1): Operation not permitted
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
```
So the thing to check is the kernel log around the same time (either with `dmesg` or `sudo journalctl --since="10:53:00" --until="10:53:30"`)

cryd-s, 2023-03-19:
```
-- Journal begins at Sat 2023-03-18 17:20:36 CET, ends at Sun 2023-03-19 11:02:48 CET. --
Mar 19 10:53:00 motioneye meyectl[10488]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10488]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10490]: mount error(22): Invalid argument
Mar 19 10:53:00 motioneye meyectl[10490]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10492]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10492]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10494]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10494]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10496]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10496]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[10498]: mount error(1): Operation not permitted
Mar 19 10:53:00 motioneye meyectl[10498]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:00 motioneye meyectl[109]:    ERROR: failed to mount smb share "//192.168.178.153/nvr" at "/media/motioneye_192_168_178_153_nvr_administrator"
Mar 19 10:53:00 motioneye meyectl[109]:    ERROR: access to network share //192.168.178.153/nvr failed: cannot mount network share
Mar 19 10:53:15 motioneye sudo[10480]: pam_unix(sudo:session): session closed for user root
Mar 19 10:53:16 motioneye sudo[10503]:     root : TTY=pts/3 ; PWD=/root ; USER=root ; COMMAND=/usr/bin/journalctl -xu motioneye
Mar 19 10:53:16 motioneye sudo[10503]: pam_unix(sudo:session): session opened for user root(uid=0) by root(uid=0)
Mar 19 10:53:27 motioneye meyectl[10506]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10506]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10508]: mount error(22): Invalid argument
Mar 19 10:53:27 motioneye meyectl[10508]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10510]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10510]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10512]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10512]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10514]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10514]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[10516]: mount error(1): Operation not permitted
Mar 19 10:53:27 motioneye meyectl[10516]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
Mar 19 10:53:27 motioneye meyectl[109]:    ERROR: failed to mount smb share "//192.168.178.153/nvr" at "/media/motioneye_192_168_178_153_nvr_administrator"
Mar 19 10:53:27 motioneye meyectl[109]:    ERROR: access to network share //192.168.178.153/nvr failed: cannot mount network share
```

zagrim, 2023-03-19:
It might be that the permissions of the mount point are off, or the settings in `/etc/fstab` for the mount.

cryd-s, 2023-03-19:
[image]

cryd-s, 2023-03-19:
[image]

zagrim, 2023-03-19:
Oh, so you are running this in a container (docker or something like that)? I'm sorry but I cannot recognise from the screenshots the environment.
In that case you should probably handle mounting of the SMB share outside of the container and just give access to it to the container.

cryd-s, 2023-03-19:
i run it in a lxc container on proxmox and my file storage is a debian server lxc container

zagrim, 2023-03-19:
Would this have something that helps you out: https://unix.stackexchange.com/questions/449960/why-do-i-get-mount-error1-operation-not-permitted-on-mount-cifs-in-a-lxc

cryd-s, 2023-03-19:
[image]

cryd-s, 2023-03-19:
[image]

to connect and create files from motioneye to fileserver is no problem with this but the mount doesnt work :(

zagrim, 2023-03-19:
I was referring to this in particular, and the linked contents from there on: https://unix.stackexchange.com/a/449961

So, one possible solution would be to make the container privileged. How to do that, you have to find out yourself since I do not know. But security-wise it would probably be better to follow the longer path: https://bayton.org/docs/linux/lxd/mount-cifssmb-shares-rw-in-lxd-containers/

cryd-s, 2023-03-19:
Ok useing a privileged Container fixed the Problem

closed 2023-03-19 by cryd-s