motioneye-project/motioneye #2745: Motioneye freezes when adding a camera

tperich, 2023-04-08:
Using docker image `ccrisan/motioneye:master-amd64`, the exact `docker-compose.yml` is:

```
version: '3'
services:
  motioneye:
   image: ccrisan/motioneye:master-amd64
   container_name: motioneye
   restart: unless-stopped
   user: root
   ports:
     - 8765:8765
   devices:
     - /dev/video1:/dev/video0
     - /dev/video3:/dev/video1
   volumes:
     - "/etc/localtime:/etc/localtime:ro"
     - "./etc:/etc/motioneye"
     - "./lib:/var/lib/motioneye"
```

The problem arises when adding a camera from the web interface. 

When clicking the dropdown for v4l2 cameras and selecting one, the web interface sends a request:

```
POST /config/add/?_=1680983613857&_username=admin&_signature=598ac07daca4809ea13e9b6fa9b7cd20e0ef9a39 undefined
Host: 192.168.1.12:8765
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:109.0) Gecko/20100101 Firefox/111.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Content-Type: application/json
X-Requested-With: XMLHttpRequest
Content-Length: 37
Origin: http://192.168.1.12:8765
Connection: keep-alive
Referer: http://192.168.1.12:8765/
Cookie: meye_username=_
DNT: 1
Sec-GPC: 1
```

and the server never responds, it just keeps hanging.

The last log produced by the container is:
`motioneye  |     INFO: adding new v4l2 camera with id 1...`

Killing the server and starting again breaks it indefinitely. The server just resets the connection upon each request.

Is there a newer docker image? Is there a better way to debug this?

rothn, 2023-10-31:
I'm having the same issue

JamBalaya56562, 2026-11-12:
`ccrisan/motioneye:master-amd64` is the old Python 2 image, which is no longer built; the current one is `motioneyeproject/motioneye` (see `docker/README.md`). As there was no follow-up since 2023, I am closing this. If adding a camera still hangs with the current image, please open a new issue with the motionEye log at debug level.

closed 2026-11-12 by JamBalaya56562