motioneye-project/motioneye #2746: Request for change to installation doc: pillow on Bullseye

adashrod, 2023-04-09:
I recently installed motioneye on a Raspberry Pi Zero W and ran into issues installing python pillow. On [step 4](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye), the installation stopped at 99% and never finished, but the alternative method (`apt-get install python-pil`) didn't work because the package doesn't exist in the official repo (`E: Unable to locate package python-pil`).

### Suggestion:

Add these steps as the alternative if `pip2 install motioneye` fails to install pillow:

```bash
wget https://github.com/python-pillow/Pillow/archive/refs/tags/6.2.2.tar.gz
pip2 install 6.2.2.tar.gz
```

MichaIng, 2023-04-10:
Check out the method for the new Python 3 compatible version: https://github.com/motioneye-project/motioneye/tree/dev#installation

JamBalaya56562, 2026-10-27:
The per-distribution installation pages have been removed from the wiki (December 2025). There is now a single install method for all distributions in the README: https://github.com/motioneye-project/motioneye#installation. It is for Python 3, and it lists the build dependencies needed to compile Pillow on ARMv6 boards like the Pi Zero W. Closing, as the page this refers to no longer exists. If the README instructions do not work for you, please open a new issue.

closed 2026-10-27 by JamBalaya56562