motioneye-project/motioneye #2747: Motioneye using huge cpu

antonical, 2023-04-13:
Motioneye docker running on an EXSI host with 8 x Xeon E-2650 Cpu's all cpu's maxed at 100% all the time. Not recording. 6 cameras are connected. 

Is this normal?

Cheers
Spart

MichaIng, 2023-04-15:
It all depends on hardware acceleration, which might be not made available for VM guests. Not sure what ESXi is capable of but check whether you find and enable GPU acceleration settings for VMs.

Also use tools like `htop` to find our which exact process is causing this high load.

Should be the case, but might be worth to test whether the bare metal install (instead of Docker) works better. A Python module should be trivial enough to install so that Docker IMO has not really any benefit, as long as you do not have a specific reason to require the additional sandboxing.

JamBalaya56562, 2026-11-12:
Closing, as there was no reply to MichaIng's suggestions: make GPU acceleration available to the VM, find the busy process with `htop` (usually `motion`, which decodes each camera stream), and try a non-Docker install. Using the cameras' substreams also lowers the load. Please open a new issue if the load is still high on a current release.

closed 2026-11-12 by JamBalaya56562