motioneye-project/motioneye #2750: Grey screnn

MiNuSMaiD, 2023-04-24:
Hello,
I have installed cctv 4k resolution 3840*2160 cameras from several brands on my installation (hikvision, ubiquiti, ankee) I add the cameras well, communication is going well. the problem is that the image shows different gray images or clouds of gray images. do you have a solution so that it is more gray image appearance? to know I have lowered my framerate 5i/s and it does not change anything
![20230422_230204.jpg](https://user-im[image]

3977775-f341ca3f-3ecb-4b5b-b4d1-e8e2404c1630.jpg)

[image]

[image]

[image]

jwhite, 2023-05-10:
I have a similar issue. It isn't constant, but every minute or so it goes grey.

VLC can decode the rstp stream correctly.  This is an ezviz outdoor camera.  

The codec listed in vlc is Codec: MPEG-H Part2/HVEC (H.265) (hevc)

jwhite, 2023-05-10:
[image]

MiNuSMaiD, 2023-05-11:
It is the dev version v0.42.1 of motionEye. MotionEye is running on a Xeon processor with virtualization overlay under Proxmox, with a 10GB network card and 32GB of RAM

MichaIng, 2023-05-13:
Could be tested with the new version. I do not remember a fix explicitly done about such issue, but there have been a lot of commits I am not aware of, respectively done before I joined the project: https://github.com/motioneye-project/motioneye/tree/dev#installation

JamBalaya56562, 2026-11-12:
Closing, as there was no report from a current release after MichaIng's question. Grey or smeared frames come from Motion decoding the stream: at about 2 fps here with 4K H.265, the decoder does not keep up or loses packets. A camera substream with a lower resolution, or H.264 instead of H.265, usually helps. Please open a new issue with the motionEye and Motion versions if it still happens on a current release (https://github.com/motioneye-project/motioneye#installation).

closed 2026-11-12 by JamBalaya56562