motioneye-project/motioneye #2751: Failure to start when port is set to 80

quint21, 2023-04-24:
Issue: When the TCP port is set to the default of 8765, or almost any other port that I've tested, it will start as normal. But, when you change "the TCP port to listen on" to port 80, MotionEye 0.43.0 will not start.

I have tested this on two different Debian-based distros that I have used extensively with Motioneye (always running on port 80) in the past, Openmediavault v6, and DietPi 8.15.2. In both cases, MotionEye was installed and is run, by the root user (which, to my understanding, should allow MotionEye to use port 80?)

No other services are using port 80. (I have OMV's management console configured to use port 81. DietPi is a minimal distro that does not have any servers running by default.)

The legacy MotionEye, 0.42.1, has no trouble running on port 80 on the same systems. Is this a Python 2 vs Python 3 thing? Is it a Linux kernel thing? Or, is it something to do with the new codebase? Thanks in advance!

systemctl status message follows:

`root@DietPi:/etc/motioneye# systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; disabled; vendor preset: enabled)
     Active: inactive (dead)

root@DietPi:/etc/motioneye# systemctl start motioneye
root@DietPi:/etc/motioneye# systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2023-04-24 20:09:40 BST; 905ms ago
    Process: 656 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 656 (code=exited, status=1/FAILURE)
        CPU: 1.586s

Apr 24 20:09:40 DietPi meyectl[656]:     sock.bind(sockaddr)
Apr 24 20:09:40 DietPi meyectl[656]: PermissionError: [Errno 13] Permission denied
Apr 24 20:09:40 DietPi meyectl[656]: Exception ignored in: <function Pool.__del__ at 0x7f480a21adc0>
Apr 24 20:09:40 DietPi meyectl[656]: Traceback (most recent call last):
Apr 24 20:09:40 DietPi meyectl[656]:   File "/usr/lib/python3.9/multiprocessing/pool.py", line 268, in __del__
Apr 24 20:09:40 DietPi meyectl[656]:   File "/usr/lib/python3.9/multiprocessing/queues.py", line 372, in put
Apr 24 20:09:40 DietPi meyectl[656]: AttributeError: 'NoneType' object has no attribute 'dumps'
Apr 24 20:09:40 DietPi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Apr 24 20:09:40 DietPi systemd[1]: motioneye.service: Failed with result 'exit-code'.
Apr 24 20:09:40 DietPi systemd[1]: motioneye.service: Consumed 1.586s CPU time.
`

MichaIng, 2023-04-25:
This is a `motion` user vs `root` user thing. For security reasons, as common best practice, motionEye does not run as `root` user anymore. Do this to grant the service permissions to bind to ports below 1024:
```sh
mkdir -p /etc/systemd/system/motioneye.service.d
echo -e '[Service]\nAmbientCapabilities=CAP_NET_BIND_SERVICE' > /etc/systemd/system/motioneye.service.d/port80.conf
systemctl daemon-reload
systemctl restart motioneye
```
If someone finds time, a page about this in out Wiki would be nice: https://github.com/motioneye-project/motioneye/wiki

For reference:
- https://www.freedesktop.org/software/systemd/man/systemd.exec.html#AmbientCapabilities=
- https://manpages.debian.org/capabilities#CAP_NET_BIND_SERVICE (applies to all Linux distros)

The service could be also adjusted to run as `root` user again, but I strongly recommend to NOT do this for any web service which does not at minimum drops its own privileges for the actual request handlers (like common web servers, but motionEye does not do that).

JamBalaya56562, 2026-11-12:
MichaIng's answer above is the fix: since the Python 3 releases, the service runs as the `motion` user (`User=motion` in `motioneye/extra/motioneye.systemd`), which cannot bind to ports below 1024. The drop-in with `AmbientCapabilities=CAP_NET_BIND_SERVICE` allows port 80 without running motionEye as root. Closing as answered.

closed 2026-11-12 by JamBalaya56562