motioneye-project/motioneye #2755: motion detection mask changes

janlammertyn, 2023-05-04:
I have a weird problem. The motion detection mask I use changes over time. It is like shifting downwards. For example, on the grid, I counted the number of masked blocks and open blocks in the first column. This is the result:

t0: 26 masked, 4 open, 12 masked
t1: 27 masked, 4 open, 11 masked
t2: 28 masked, 3 open, 11 masked
t3: 30 masked, 3 open, 9 masked.

I did not edit the mask in between.

MotionEye v0.43.0
Motion v 4.5.1
OS Ubuntu 22.04

MichaIng, 2023-05-04:
Probably somehow related to #2404. But the mask file is actually stored only when saving camera settings, AFAIK, so not sure how it can permanently change. It is definitely not related to a changing camera view width/height, e.g. due to changing browser window width or similar? Does it reset when you restart motionEye?

However, as of my findings in the linked link, the whole mask code needs a cleanup.

srmeister, 2023-06-12:
yeah i have the same problem.
It seems to move down and to the right 1 unit when changing anything. if i had to guess, there is error either when saving or reading the mask.

zagrim, 2023-07-05:
I tried to reproduce this with current dev branch with Motion v 4.5.1 by changing settings that don't affect the size or position of the video feeds (and thus I could easily check if the mask had moved), but couldn't get the mask to drift. It might of course be triggered by camera resolution (or aspect ratio), and mine produce different resolutions than @srmeister has, possibly they also differ from what @janlammertyn has.

Marijn0, 2024-07-23:
I can replicate this issue if the Resolution Dimmer is not set to 100%. Here is an example with the Resolution Dimmer set to 80%:

before:
[image]
After:
[image]

Resolution Dimmer 5%:
[image]

I also noticed the following in the log files when this happens,
journalctl -u motioneye:
DEBUG: editable mask needs resizing from 819x614 to 1024x768

zagrim, 2025-01-19:
Ok, now I managed to somewhat replicate the mask drift (I guess the amount of drift is related to how nicely the dimming factor divides the original resolution, or something like that). But it is still too small to be easily checked, so it would be great if someone could provide steps to reproduce this in the scale reported here.

For some reason I also get a flood of repeated logging like below when enabling resolution dimmer, so I can't really use debug logging to inspect this :disappointed: 
```
   DEBUG: tag: DateTime (306) - type: string (2) Tag Location: 22 - Data Location: 80 - value: b'2025:01:19 14:54:28\x00'
   DEBUG: tag: ExifIFD (34665) - type: long (4) - value: b'\x00\x00\x002'
   DEBUG: tag: unknown (34858) - type: signed short (8) - value: b'\x00\x02'
```

Anyway, the mask files do get re-saved (and after that re-read) every time camera settings are changed regardless whether the mask has been edited, so any rounding errors etc can easily accumulate over time.

zagrim, 2025-01-19:
I wonder if it would be possible to remedy this by using the `resolutionFactor` and the real camera frame `width` and `height` in code to do the required scaling instead of depending on the DOM element dimensions when parsing/building the mask 🤔 However, as the mask gets read and saved repeatedly when settings are changed it's inevitable that any rounding errors will cause some drift.

I'm not quire sure if this is real, but having the mask displayed in UI affect the mask saved thereafter is quite problematic. I mean that the fact that the UI might have different frame resolution than what the camera actually produces seems to now "leak" to the mask used by Motion with the real camera video. Maybe the UI should display a copy of the mask that is in the "viewable resolution" but keep the unaltered one in store so it can be used when saving settings?

zagrim, 2025-01-19:
Ah, I guess what is needed to reproduce this is to change the resolution dimmer and then go edit the mask, right?

zagrim, 2025-01-19:
Like MichaIng mentioned above, it seems that the mask code indeed needs some work also for reasons other than the one mentioned here. I got horrific distortion in the grid causing weirdly sized mask tiles in the bottom and right edges when playing around with the resolution dimmer between mask edits 😬

marbetschar, 2025-06-15:
Just to add another data point to the discussion here:

I was also experiencing a drift of both masks (privacy and motion detection mask). The resolution dimmer is set to 100% on my end, so I started tinkering with the video resolution.

Playing with the video resolution revealed that sometimes both masks would drift either down- or upwards. I then checked the supported resolutions of my camera using v4l2-ctl:

```
$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Stepwise 16x16 - 16376x16376 with step 1/1
	[1]: 'UYVY' (UYVY 4:2:2)
		Size: Stepwise 16x16 - 16376x16376 with step 1/1
        ...
```

There are many more entries shown - but curiously all of them contain `Size: Stepwise 16x16 - 16376x16376`. That said, I started to experiment with `Video resolution: Custom` in motionEye and used only multiples of 16 for width and height. This fixed the drift of both masks on my end.

MichaIng, 2026-10-22:
Issue not yet solved if a resolution dimmer is used.

closed 2026-10-31 by MichaIng