motioneye-project/motioneye #2756: Dropbox access key expires after 4 hours

SLongofono, 2023-05-06:
Using the latest raspberry pi image as of 2023-05-05 with dropbox, targeting the folder /camera.

The first day, it successfully requested a token, I approved it manually in my browser, and then applied settings in the motioneye web interface.  Videos were uploading correctly.

Approximately 5 hours later, the uploads stopped working.  Initiating another "Test Service" from the web interface shows that the key is expired.

Is there a way to auto-refresh these keys or adjust the settings such that they don't need to be refreshed several times a day?

MichaIng, 2023-05-06:
Do you already use the Python 3 version of motionEye installed following these instructions? https://github.com/motioneye-project/motioneye/tree/dev#installation

Else: What do you mean with "latest raspberry pi image"? Raspberry Pi OS, or motionEyeOS? The letter still ships with the old motionEye version, which is known to not work well with Dropbox anymore, while the new version has this fix added: #2396

SLongofono, 2023-05-06:
@MichaIng My mistake, didn't notice that this repo was distinct from the motionEyeOS one.  This was the latest image of motionEyeOS, is there a way to patch that or will I need to start from scratch?

MichaIng, 2023-05-10:
Due to it's container like nature, you cannot really update motionEye within motionEyeOS but we would need to update motionEyeOS. However, currently we struggle to find enough time to update motionEye itself (and do a new stable release), which is the first step.

You can try some other OS like Raspberry Pi OS, or DietPi (which I maintain myself) and install motionEye via above instructions.

JamBalaya56562, 2026-11-12:
As MichaIng explained, the motionEyeOS image still ships the old motionEye, and the Dropbox fix (#2396) is in the Python 3 releases of motionEye. On another OS, the current release can be installed with the steps in the README (https://github.com/motioneye-project/motioneye#installation); an update of motionEyeOS itself belongs to the motionEyeOS repository. Closing as answered.

closed 2026-11-12 by JamBalaya56562