motioneye-project/motioneye #2759: Videos are not saved, neither with motion detection nor normal recording

backupandforkacc, 2023-05-12:
The settings seem ok. I can see the stream online, but no video is saved. Neither in the file on the filesystem nor through the explorer in the frontend of motioneye I can see any recorded videos. I can see the pictures when setting to normal recording, but not with motion detection. The motion detection also triggers, at least I get messages via telegram.

[image]
[image]

[camera-1.conf.txt](https://github.com/motioneye-project/motioneye/files/11467075/camera-1.conf.txt)
[motion.conf.txt](https://github.com/motioneye-project/motioneye/files/11467076/motion.conf.txt)
[motioneye.conf.txt](https://github.com/motioneye-project/motioneye/files/11467077/motioneye.conf.txt)

MarcelCoding, 2023-05-13:
I can confirm, since I switched to the `edge` docker tag, videos aren't getting saved anymore.

MichaIng, 2023-05-13:
Thanks for reporting. Is the container reporting any errors messages? I think with plain motionEye installed via `pip` it works 🤔.

backupandforkacc, 2023-05-13:
Sorry, I now indeed have some error, which I have not detected yesterday:

/var/log/motioneye/motion.log shows

```
[1:ml1:IMOU] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
[1:ml1:IMOU] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
[1:ml1:IMOU] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
```
Also ffmpeg seems to be ok
```
 ffmpeg -v
ffmpeg version 4.3.6-0+deb11u1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
```

backupandforkacc, 2023-05-14:
I just installed from scratch as [explained here](https://github.com/motioneye-project/motioneye/wiki/Install-on-Debian-11-(Bullseye))
and still get the error:
```

[1:ml1:Camera1] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
[1:ml1:Camera1] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
[1:ml1:Camera1] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
```

MarcelCoding, 2023-05-18:
Today I got it working using mp4 without omx.

JamBalaya56562, 2026-11-12:
The encoder in your log was the default movie format: new cameras got `mp4:h264_omx` whenever ffmpeg listed that encoder, even where the hardware cannot open it, as on Bullseye with the KMS driver. Since #3543, new cameras get plain H.264 (`mp4`) instead; the OMX and V4L2 M2M encoders are still selectable. A camera that already has `H.264/OMX` keeps it, so switch it to `H.264 (.mp4)` under Movies, as MarcelCoding did. Closing as fixed by #3543.

closed 2026-11-12 by JamBalaya56562