motioneye-project/motioneye #2762: Impossible to add a new network camera

dukei, 2023-05-18:
Trying to add a new network camera after a fresh install on Ubuntu (`sudo pip install motioneye.eo`) results in error in syslog:
```
May 18 15:10:13 ubuntus meyectl[366423]:    ERROR: Exception in callback test_mjpeg_url.<locals>.on_response(<Future finis...was received.>) at /usr/local/lib/python3.8/dist-packages/motioneye/utils/mjpeg.py:117
May 18 15:10:13 ubuntus meyectl[366423]: handle: <Handle test_mjpeg_url.<locals>.on_response(<Future finis...was received.>) at /usr/local/lib/python3.8/dist-packages/motioneye/utils/mjpeg.py:117>
May 18 15:10:13 ubuntus meyectl[366423]: Traceback (most recent call last):
May 18 15:10:13 ubuntus meyectl[366423]:   File "/usr/lib/python3.8/asyncio/events.py", line 81, in _run
May 18 15:10:13 ubuntus meyectl[366423]:     self._context.run(self._callback, *self._args)
May 18 15:10:13 ubuntus meyectl[366423]:   File "/usr/local/lib/python3.8/dist-packages/motioneye/utils/mjpeg.py", line 119, in on_response
May 18 15:10:13 ubuntus meyectl[366423]:     if response.code == 401 and auth_modes and url_obj.username:
May 18 15:10:13 ubuntus meyectl[366423]: AttributeError: '_asyncio.Future' object has no attribute 'code'
```

Versions installed:
```
user@ubuntus:~$ sudo pip install motioneye.eo --upgrade
Requirement already up-to-date: motioneye.eo in /usr/local/lib/python3.8/dist-packages (22.4.1)
Requirement already satisfied, skipping upgrade: babel in /usr/lib/python3/dist-packages (from motioneye.eo) (2.6.0)
Requirement already satisfied, skipping upgrade: pycurl in /usr/lib/python3/dist-packages (from motioneye.eo) (7.43.0.2)
Requirement already satisfied, skipping upgrade: boto3 in /usr/lib/python3/dist-packages (from motioneye.eo) (1.9.253)
Requirement already satisfied, skipping upgrade: pillow in /usr/lib/python3/dist-packages (from motioneye.eo) (7.0.0)
Requirement already satisfied, skipping upgrade: tornado in /usr/lib/python3/dist-packages (from motioneye.eo) (5.1.1)
Requirement already satisfied, skipping upgrade: jinja2 in /usr/lib/python3/dist-packages (from motioneye.eo) (2.10.1)
```

MichaIng, 2023-05-21:
This is the wrong repository for the module you installed. You'd go here: https://github.com/jmichault/motioneye.eo
However, the two have been merged. So I suggest to install the `dev` branch form this repo which should have a bunch of additional features and fixes: https://github.com/motioneye-project/motioneye/tree/dev#installation

free-cup-holder-exe, 2026-11-08:
The traceback comes from an old copy of `mjpeg.py` in the `motioneye.eo` package; `dev` has not had that one since 2022. What remained of the problem, the add dialog waiting when that check fails, is handled by #3443 in 0.45.0b1: an `http://` camera that can't be used shows the reason within seconds instead.

free-cup-holder-exe, 2026-11-08:
Closing as completed; please reopen if adding a network camera still fails like this with 0.45.0b1.

closed 2026-11-08 by free-cup-holder-exe