motioneye-project/motioneye #2763: Pi Zero W and Camera Module 3

himuura, 2023-05-18:
Well, for the life of me, i cannot manage to put the camera module 3 to work with a pi zero w and motioneye latest stable version. it says unable to connect to device (works fine with raspbian os though). tried addind it as mmal cam, not detected...vcgen command says supported but not connected. Any ideas?

himuura, 2023-05-18:
Camera Module 3 is not yet supported...to install and run MotionEye, you need to follow these instructions https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye
Needless to say, it's very slow on a pi zero w with OC (1100 Mhz). it does the work some times but motion detection for video and photos it's a pain to say the least, CPU usage at 100% most of the times. Gonna try with a pi 2 W to see if it works better (should work).

strips, 2023-05-21:
See here for issues with compatibility and RPI cam v3 https://github.com/motioneye-project/motioneye/issues/2683.

MichaIng, 2023-05-21:
Better install the new version via these instructions: https://github.com/motioneye-project/motioneye/tree/dev#installation

JamBalaya56562, 2026-11-03:
The Camera Module 3 works with current motionEye through `libcamerify`, following the steps in #2812: checked on a Raspberry Pi 5 with a Camera Module 3 and motion 4.7.1, not on a Pi Zero W. Native libcamera support is tracked in #2812, so closing this as a duplicate of it. If the camera still does not work with these steps, please reopen.

closed 2026-11-03 by JamBalaya56562