motioneye-project/motioneye #2764: MotionEye does not work any more after reboot, only first setup working

Denn2705, 2023-05-18:
Hello all,

im trying to get MotionEye running on my Pi3B. I have set Up Debian 11 Lite and followed the installation steps written here:
https://github.com/motioneye-project/motioneye/tree/dev#installation

When im doing this, my MotionEye and Camera are working well. But as soon as i reboot my Pi, i cannot access the Website any more. The IP-Adresse:8765 shows me always "Page cannot be loaded, Connection failed"

What did i miss? Is there something missing in the Autostart? Or can i check what component is not running probably after reboot?

Thanks in advance!
Br
Dennis

MichaIng, 2023-05-21:
Please check the service logs:
```sh
journalctl -u motioneye
```

xXSkyWalkerXx1, 2024-03-17:
I have no entries there and the container prints only 'INFO: hello! this is motionEye server 0.42.1' (after reboot).
On first boot everything works fine.

xXSkyWalkerXx1, 2024-03-17:
> I have no entries there and the container prints only 'INFO: hello! this is motionEye server 0.42.1' (after reboot).
> On first boot everything works fine.

I have no entries there and the container prints only 'INFO: hello! this is motionEye server 0.42.1' (after reboot).
On first boot everything works fine.

MichaIng, 2024-03-17:
v0.42.1 is the old version, and the instructions you linked would usual v0.43.1b1 and not a container but bare metal. Probably you have two conflicting motionEye instances now, bare metal v0.43.1b1 and container v0.42.1.

Latest version containers can be found here: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

JamBalaya56562, 2026-11-12:
MichaIng's answer above covers this: `journalctl -u motioneye` shows why the service did not come up after the reboot, and a container and a bare metal install should not run side by side. As there was no reply with the log, I'm closing this. If it still happens on the current version, please open a new issue with that log.

closed 2026-11-12 by JamBalaya56562