motioneye-project/motioneye #2766: Thumbnail offset hardcoded weirdly?

suvis1, 2023-05-20:
Hi!

Ive been having issues that Motioneye generates thumbnails that are not representative of the actual detection event. I checked the code and found out theres a a minimum offset, and also a weirdly doubled offset for precapture?

For example if i have a camera with framerate of 10, and precapture of 40 frames before a detection, we end up with the thumbnail being generated 8 seconds from the video, eventhough the detection is after the 40 precapture frames, meaning the thumbnails 4 seconds late.

I managed to get alot better thumbnails when i changed the offs value to just be offs +=1 so the thumbnails are taken 1 second after detection.

This seems to be done deliberately so maybe im missing the point of this? is there any reason for it to do this? 

https://github.com/motioneye-project/motioneye/blob/03882a18994fc35c22ae169dd1569abb923c7cef/motioneye/mediafiles.py#L349-L363

MichaIng, 2023-05-21:
Many thanks for your report. Good find, I have no idea where this 4 seconds minimum + doubled offset is coming from. It was last touched here, before which it was 4 seconds maximum offset: https://github.com/motioneye-project/motioneye/commit/b14bbb5

@ccrisan do you know what this is for? Generally I'd just remove any min/max and the `* 2` to just use the user setting as is.

suvis1, 2023-05-23:
I feel like some delay is still warranted, since for example a thumbnail taken from the moment of detection only shows a very small part of a detection like this:

[image]
I blurred it for privacy, but you can still see that only the head of the person is in the thumbnail on the driveway, adding 1 second more generally gives a lot better results!

[image]

Semmu, 2026-01-29:
I have recently also dig into this code and IMHO the best solution would be to create the thumbnail from the exact middle of the video, I think that would be more informative than the beginning of the motion detection/event.